ghdl/ghdl #2555: GHDL reports `incomplete sensitivity list` when `rising_edge` is enclosed in parenthesis

pidgeon777, 2023-12-07:
This code returns a warning:

```vhdl
PROCESS_p: process(CLK)
begin
  if (rising_edge(CLK)) then
	sig <= INPUT_P;
  end if;
end process;

```

Warning message:

```log
incomplete sensitivity list, signal "INPUT_P" is missing
```

but if I remove the parenthesis enclosing `rising_edge`, the warning disappears.

**Context**

- OS: Windows 10 Pro
- GHDL version:

```log
GHDL 4.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2023-12-07:
Do you have a full reproducer (including the options you have used) ?

Paebbels, 2024-12-19:
@pidgeon777 can you provide a reproducer?