ghdl/ghdl #2556: [BUG] GHDL gets stuck when stopping simulation

pidgeon777, 2023-12-07:
**Description**

Consider this code:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture tb of test is
begin
   p_main: process
   begin
    std.env.stop;
    wait;
  end process p_main;
end architecture;
```

And these commands:

```powershell
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 test
```

after starting the simulation, `GHDL` reports:

```log
simulation stopped @0ms
```

but the problem is that `GHDL` process gets stuck, and doesn't exit until a good amount of time (1 minute or so).

Sometimes, when called inside my `nvim` editor as a compilation task, it never exit.

This very same behaviour occurs when using `report failure` statements, with or without `asserts`.

**Expected behaviour**

I would expect `GHDL` to immediately quit itself after simulation stop condition is met.

Maybe this issue is related to the following?:

https://github.com/ghdl/ghdl/issues/2543

**Context**

- OS: Windows 10 Pro
- GHDL version:

```log
GHDL 4.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2023-12-07:
Yes, it's the same issue.  mcode is not fully supported on win64.

avelure, 2023-12-13:
In other words, for the time being, from the nightly you can use 
[mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst](https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst)
instead of 
[mingw-w64-ucrt-x86_64-ghdl-mcode-ci-1-any.pkg.tar.zst](https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-ucrt-x86_64-ghdl-mcode-ci-1-any.pkg.tar.zst)

pidgeon777, 2024-02-11:
@avelure or @tgingold what should be the difference between those two archives you suggested?

I see there is `mcode` writte in both?

avelure, 2024-02-12:
x86_64 is compiled for 64bit, i686 is compiled for 32bit

tgingold, 2024-04-10:
Win64 is now fully supported with mcode.

commit 76ab759c (2024-04-14): testsuite/gna: add a test for #2556

Close #2556
closed 2024-04-14 by tgingold