ghdl/ghdl #2557: Significant slowdown of analyzis

wzab, 2023-12-08:
**Description**
The problem below occurs in GHDL compiled with LLVM. I haven't checked other versions.
In one of our designs we use a big ROM storing structured data generated with the Python script (I attach the archive containing the generated VHDL file and packages it uses). The file pout.vhd was analysed very quickly with GHDL 3.0.0.
Unfortunately, the current master version is not able to analyze that file in reasonable time.

**Expected behaviour**
Analysis of three provided files:

    ghdl -a -g -C -O2 --std=08 --ieee=synopsys omtf_constants.vhd
    ghdl -a -g -C -O2 --std=08 --ieee=synopsys golden_proc_pkg.vhd
    ghdl -a -g -C -O2 --std=08 --ieee=synopsys pout.vhd

should be completed in a few seconds. That's how it is done with GHDL 3.0.0.

**How to reproduce?**

Unpack the attached archive [mwe.zip](https://github.com/ghdl/ghdl/files/13617379/mwe.zip).
Run the script `./test.sh` with GHDL 3.0.0 in PATH (the script should be executed quickly)
Remove the produced *.o and *.cf files
Run the script `./test.sh` with newest GHDL in PATH. The analysis will be very slow.

**Context**
Please, provide the following information:

- OS: Debian/testing
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ *] Built from sources: `09b9dd7aa89f418ca809587e4a440e266a218cf2`

**Additional context**

tgingold, 2023-12-08:
Interesting, but I cannot reproduce it.
It takes less than 3 seconds to run your script.  So, there is something weird.

Which version of ghdl are you using (`ghdl -v` also displays the llvm version) ?
It is the same if you remove `-O2` and/or `-g` ?  You can remove optimization only for this file, as it shouldn't have a large impact.

wzab, 2023-12-08:
I have prepared another MWE that clones GHDL, checks out the particular commit, builds and installs it, and finally tries to analyze my example. It hangs on Debian/testing.
[mwe2.zip](https://github.com/ghdl/ghdl/files/13619805/mwe2.zip)

tgingold, 2023-12-08:
I am still interested in the output of `ghdl -v` to know which llvm version you are using!

tgingold, 2023-12-08:
Thanks.  I will try with llvm16.
Meanwhile, can you also try with `-O0` ?

wzab, 2023-12-08:
I have returned to the mwe2.zip and checked various optimizations.
Without -g and -O it still hangs. With -O0 it works.

tgingold, 2023-12-08:
I have just tested with llvm-17, and it works.
If the problem happens only with llvm-16, I am not sure it is worth fixing.

wzab, 2023-12-08:
OK. I'm forcing installation of llvm-17 in my machines. However, it requires use of:
`../configure --with-llvm-config=/usr/lib/llvm-17/bin/llvm-config --prefix=PREFIX` on Debian testing.
The test is in [mwe5.zip](https://github.com/ghdl/ghdl/files/13620304/mwe5.zip). Indeed it works.
Thanks!

wzab, 2023-12-08:
So the problem is fixed, unless it returns in yet newer version of LLVM...

closed 2023-12-10 by tgingold