ghdl/ghdl #2559: Failure to convert bit vector literal into string using `to_hstring`.

jeras, 2023-12-12:
Using GHDL from git:
```sh
$ ghdl --version
GHDL 4.0.0-dev (3.0.0.r714.gf171833e7) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Output on attempt:

```sh
../axi/AXI4-Stream/tb/CslTbAxiStreamPkg.vhd:95:46:error: can't resolve overload for function call, slice or indexed name
         report "DEBUG: data = " & to_hstring(8X"a5");
                                             ^
../axi/AXI4-Stream/tb/CslTbAxiStreamPkg.vhd:95:46:error: possible interpretations are:
../../src/ieee2008/numeric_std.vhdl:1607:12:error: function to_hstring [unresolved_signed return string]
../../src/ieee2008/numeric_std.vhdl:1606:12:error: function to_hstring [unresolved_unsigned return string]
../../src/ieee2008/std_logic_1164.vhdl:268:12:error: function to_hstring [std_ulogic_vector return string]
*std_standard*:1:1:error: function to_hstring [bit_vector return string]
../axi/AXI4-Stream/tb/CslTbAxiStreamPkg.vhd:95:34:error: no function declarations for operator "&"
         report "DEBUG: data = " & to_hstring(8X"a5");
                                 ^
*command line*:1:1:error: unit entity "cslaxistreamresizetb" has not been analyzed
```

tgingold, 2023-12-12:
The error looks valid.  You need to qualify the literal, like `to_hstring(bit_vector'(8x"a5"))`

closed 2023-12-16 by tgingold

jeras, 2023-12-23:
In the spirit of not ignoring null results, I em linking to an example on EDA playground.

https://www.edaplayground.com/x/kFH8