ghdl/ghdl #2563: How to pass options to GCC?

EfesX, 2023-12-17:
How can I pass options to the ghdl backend (in my case, it's gcc) using the `ghdl -i/-m` (or `ghdl -c/-e`) commands?

I mean, I want to get code coverage results using ghdl, so I want to use it like this: 
```shell
ghdl -i -Wc,-fprofile-arcs -Wc,-ftest-coverage -fsynopsys adder.vhd
ghdl -m -Wl,-lgcov adder.vhd
```
But `ghdl -i` gives an error: `ghdl:error: unknown warning identifier: c,-fprofile-arcs`.

<details>
<summary> ghdl --version </summary>

```shell
GHDL 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 GCC 9.3.0 code generator
Written by Tristan Gingold.
```
</details>

PS: I know I could use `ghdl -a/-e`, but I'm looking for a way to not concern about the design hierarchy.

tgingold, 2023-12-17:
`ghdl -i` just quickly parse the sources without any compilation.  So the `-Wx,` options are useless.

Did you try both `-Wc,xxx` and `-Wl,xxx` with `ghdl -m` ?

EfesX, 2023-12-17:
Oh, it seems to be working. Thanks.

closed 2023-12-17 by EfesX