ghdl/ghdl #2564: If-elsif-else generate with entity instantiation crashes latest GHDL master

LukasVik, 2023-12-18:
**Description**
CI using `ghdl/setup-ghdl-ci@nightly` GitHub Action started failing yesterday (Sunday) for what I think is valid VHDL code:

https://github.com/hdl-modules/hdl-modules/actions/workflows/ci.yml

https://github.com/hdl-modules/hdl-modules/actions/runs/7240913391/job/19724428663

I think I managed to distill it into an MWE below.

**Expected behaviour**
I think the code, as well as MWE, is valid VHDL so I would expect it to pass.

**How to reproduce?**

```vhd :file: mwe.vhd

library ieee;
use ieee.std_logic_1164.all;

entity mwe_entity is
  port (
    input : in std_ulogic;
    output : out std_ulogic
  );
end entity;

architecture tb of mwe_entity is

begin

  output <= input;

end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity mwe_entity2 is
  port (
    input : in std_ulogic;
    output : out std_ulogic
  );
end entity;

architecture tb of mwe_entity2 is

begin

  output <= not input;

end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity mwe_tb is
end entity;

architecture tb of mwe_tb is

  constant test_boolean : boolean := false;

  signal input, output : std_ulogic;

begin

  ------------------------------------------------------------------------------
  choose_dut : if test_boolean generate

    dut : entity work.mwe_entity
      port map (
        input => input,
        output => output
      );

  else generate

    dut : entity work.mwe_entity2
      port map (
        input => input,
        output => output
      );

  end generate;

end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity mwe_tb2 is
end entity;

architecture tb of mwe_tb2 is

  constant test_boolean : boolean := false;

  signal input, output : std_ulogic;

begin

  ------------------------------------------------------------------------------
  choose_dut : if test_boolean generate

    dut : entity work.mwe_entity
      port map (
        input => input,
        output => output
      );

  else generate

    -- dut : entity work.mwe_entity2
    --   port map (
    --     input => input,
    --     output => output
    --   );

  end generate;

end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity mwe_tb3 is
end entity;

architecture tb of mwe_tb3 is

  constant test_boolean : boolean := true;

  signal input, output : std_ulogic;

begin

  ------------------------------------------------------------------------------
  choose_dut : if test_boolean generate

    dut : entity work.mwe_entity
      port map (
        input => input,
        output => output
      );

  else generate

    dut : entity work.mwe_entity2
      port map (
        input => input,
        output => output
      );

  end generate;

end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity mwe_tb4 is
end entity;

architecture tb of mwe_tb4 is

  constant test_boolean : boolean := true;

  signal input, output : std_ulogic;

begin

  ------------------------------------------------------------------------------
  choose_dut : if test_boolean generate

    -- dut : entity work.mwe_entity
    --   port map (
    --     input => input,
    --     output => output
    --   );

  else generate

    dut : entity work.mwe_entity2
      port map (
        input => input,
        output => output
      );

  end generate;

end architecture;

```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 mwe.vhd

ghdl --elab-run --std=08 mwe_tb
ghdl --elab-run --std=08 mwe_tb2
ghdl --elab-run --std=08 mwe_tb3
ghdl --elab-run --std=08 mwe_tb4
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: ecbdbe74d

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r728.gecbdbe74d) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/lukas/
Command line:
ghdl --elab-run --std=08 mwe_tb3
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-rtis.adb:2298 access check failed
******************************************************************

```

**Additional context**

Trying the MWE locally with GHDL built from git, it fails for current master (ef2c99be2) and all the way back to commit ecbdbe74d. On 27db5f6a6 it passes for me.

tgingold, 2023-12-18:
Thank you for the issue, it is now fixed.

closed 2023-12-18 by tgingold

LukasVik, 2023-12-19:
Great, thank you so much for the quick fix @tgingold. It is all working for me now!

commit 7c4bb48d (2023-12-18): trans-rtis.adb: avoid a crash on never used entities.

Fix #2564