ghdl/ghdl #2565: [Regression] CONSTRAINT_ERROR

stnolting, 2023-12-19:
Hey,

seems like I've run into a regression error. I'm using the latest GHDL version via the provided GitHub action:

```yml
uses: ghdl/setup-ghdl-ci@nightly
```

The latest nightly release (from last night) causes a crash:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r748.gef2c99be2) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
/home/runner/work/neorv32-riscof/neorv32-riscof/sim/
Command line:
/home/runner/work/_temp/ghdl/bin/ghdl -r --std=08 --work=neorv32 neorv32_riscof_tb -gMEM_FILE=./main.hex --max-stack-alloc=0 --ieee-asserts=disable --assert-level=error -gMEM_SIZE=52056 --stop-time=4ms
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-rtis.adb:2298 access check failed
******************************************************************
```

* the GitHub workflow where this error occured: https://github.com/stnolting/neorv32-riscof/actions/runs/7243717240/job/19730953538
* the used testbench: https://github.com/stnolting/neorv32-riscof/blob/main/sim/neorv32_riscof_tb.vhd

Let me know if you need further information.

tgingold, 2023-12-19:
Maybe it is the same issue as #2564 ?  Can you try again with the head ?

stnolting, 2023-12-19:
> Maybe it is the same issue as https://github.com/ghdl/ghdl/issues/2564 ?

We do not use the if-elsif-generate construct at all. Furthermore, the same code base compiled fine with GHDL's "last week" version. I think this is quite similar to #2502 🙈😅

> Can you try again with the head ?

Sure, I'll have a look.

tgingold, 2023-12-20:
I have just tried the testbench of neorv32-riscof head (with the head of ghdl): no crash.
(I think the #2564 issue is not limited to elsif-generate, so probably it's possible to crash without using this construct.  You however need to use if-generate).

stnolting, 2023-12-20:
Works now! :+1:

But I'm curious why this is the case... 😅
Does the GHDL GitHub action uses the sources from a specific release (nightly?) or does is use the latest sources from the GHDL repository's master branch?

tgingold, 2023-12-20:
Yes, I think the ghdl github action uses a docker image which is nightly built.  So depending on your setup, your CI may use a ghdl which can be 2 days old.

Anyway, thank you for the update.

stnolting, 2023-12-21:
Thanks for clearing (and for fixing/improving)!

closed 2023-12-21 by stnolting