ghdl/ghdl #2569: Constant Record as Input Raises TYPES.INTERNAL_ERROR

NikLeberg, 2023-12-26:
**Description**
If an entity input is of a record type this input cannot be connected to an constant on instantiation of the entity.
Well, it can be connected, but then while starting the simulation exception `TYPES.INTERNAL_ERROR` is raised.

**Expected behaviour**
When connecting to the constant the simulation should not raise an exception.

**How to reproduce?**
Minimal Working Example (MWE):

```vhd :file: pkg.vhdl
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;

PACKAGE pkg IS
    TYPE test_record IS RECORD
        test : STD_LOGIC;
    END RECORD test_record;
END PACKAGE pkg;
```

```vhd :file: ent_with_record_input.vhdl
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE work.pkg.ALL;

ENTITY ent_with_record_input IS
    PORT (
        a : IN test_record
    );
END ENTITY ent_with_record_input;

ARCHITECTURE arch OF ent_with_record_input IS
BEGIN
END ARCHITECTURE arch;
```

```vhd :file: ent_use_constant.vhdl
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE work.pkg.ALL;

ENTITY ent_use_constant IS
END ENTITY ent_use_constant;

ARCHITECTURE arch OF ent_use_constant IS
    CONSTANT test : test_record := (test => '0');
BEGIN
    bug : ENTITY work.ent_with_record_input
        PORT MAP(
            a => test
        );
END ARCHITECTURE arch;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a pkg.vhdl
ghdl -a ent_with_record_input.vhdl
ghdl -a ent_use_constant.vhdl
ghdl -e ent_use_constant
ghdl -r ent_use_constant
```

**Context**
Please, provide the following information:

- OS: container `gcr.io/hdl-containers/formal` running on WSL2 on Windows 10
- Origin:
  - [x] container: `gcr.io/hdl-containers/formal`, seems to use `v3.0.0-558-g8a9eb7e25` see below _bug occured_
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (v3.0.0-558-g8a9eb7e25) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/workspaces/neorv32_soc/vhdl/ghdl_bug/
Command line:
ghdl -r --std=08 ent_use_constant
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : simul-vhdl_simul.adb:3939
******************************************************************
```

**Additional context**
To circumvent the problem, one can use an ordinary signal instead of an constant (exchange `CONSTANT` with `SIGNAL`) and the simulation works as intended. The bug also triggers if an _inline constant_ is used `PORT MAP ( a => (test => '0') );`

QuestaSim / ModelSim and Quartus do not have an issue with the provided MVE. But I of course don't know if it is actual acceptable VHDL code. GHDL should then at least not crash but display an error on elaboration, maybe?

tgingold, 2023-12-30:
I think your version of ghdl is too old and you need to update it.  I cannot reproduce the crash with the head.

NikLeberg, 2023-12-31:
Thanks @tgingold!
I was assuming that `hdl-containers` would use a fairly current version. Seems to be not the case.
Sorry for the noise. This is fixed for me. 👍

commit 9a5560a7 (2024-01-01): testsuite/gna: add a reproducer for #2569

Close #2569
closed 2023-12-31 by NikLeberg