ghdl/ghdl #2570: assume => "GHDL Bug occurred"

Topi-ab, 2023-12-27:
**Description**
ghdl -a --std=08 test.vhdl
triggers GHDL Bug.

**Expected behaviour**
To pass analysis.

The first assume doesn't fail, the second fails (if commented out => no problem).

**How to reproduce?**
ghdl -a --std=08 test.vhdl

Version is images: 
- hdlc/formal:all 
- ghdl/ghdl:bullseye-mcode

As they were on Dec 27, 2023

```vhd :file: test.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity test is
  port(
    clk_in: in std_logic;
    sig_1_in: in std_logic;
    sig_2_in: in integer range 0 to 15;
    sig_3_in: in std_logic
  );
end;

architecture formal of test is
begin
  default clock is rising_edge(clk_in);

  not_failing_assume: assume {
    sig_1_in = '1' and sig_2_in = 4 and sig_3_in = '0';
    sig_1_in = '1' and sig_2_in = 5 and sig_3_in = '0';
    sig_1_in = '0' and sig_2_in = 6 and sig_3_in = '0';
    sig_1_in = '1' and sig_2_in = 7 and sig_3_in = '0'
    };

  failing_assume: assume {
    sig_1_in = '1' and sig_2_in = 4 and sig_3_in = '0';
    sig_1_in = '1' and sig_2_in = 5 and sig_3_in = '0';
    sig_1_in = '0' and sig_2_in = 6 and sig_3_in = '0';
    sig_1_in = '1' and sig_2_in = 7 and sig_3_in = '0';
    sig_1_in = '1'};

end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 test.vhdl
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (v3.0.0-558-g8a9eb7e25) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/prj/2023-12-27/
Command line:
ghdl -a --std=08 test.vhdl
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : psl-qm.adb:210 explicit raise
******************************************************************```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2023-12-30:
Yes, I can reproduce the crash.  I am investigating.

closed 2024-01-01 by tgingold

Topi-ab, 2024-01-26:
hdlc/formal doesn't have this update.

Are there changes on publishing practices?

tgingold, 2024-01-27:
I don't know how hdlc/formal produces packages, I suppose they do it from time to time, or from another source.

Topi-ab, 2024-02-02:
> I don't know how hdlc/formal produces packages, I suppose they do it from time to time, or from another source.

Year ago @umarcor triggered the update of hdl/containers

My assumption was this repo is once a week replicated to hdl/containers which should owner of hdlc/formal

https://github.com/ghdl/ghdl/issues/2392#issuecomment-1468834188

commit 2d3aed03 (2024-01-01): psl-qm: punt on too large expressions

Avoid any reduction when the input is too large (to avoid overflow).
Fix #2570