ghdl/ghdl #2571: Bug caused by guarded signal reference.

tfuto, 2023-12-28:
**Description**
A clear and concise description of what the issue is about.

**Expected behaviour**
What you expected to happen, and what is happening instead.

**How to reproduce?**

You can reproduce the erroneous behavior using test1.vhd. Note that 
 - if you set the assignment in TB to `g <= '1';` then the code runs successfully;
 - if you remove the default value of signal g, or set it to '0', then even the message "Arrived here." won't show up, and the software again shows the bug.

```vhd :file: test1.vhd
library IEEE;
use IEEE.std_logic_1164.all;

entity test1 is
end entity;

architecture beh of test1 is
  signal g : std_logic := '1';
  signal s : std_logic bus;
  
begin

  BLK : block (g = '1')
  begin
    s <= guarded '1';
  end block BLK;
  
  TB: process 
  begin
    report "Arrived here.";
    g <= '0';
    wait;
  end process TB;

end architecture beh;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a test1.vhd test2.vhd test3.vhd

# Crashes without reaching the report statement.
ghdl --elab-run test1

ghdl --elab-run test2
ghdl --elab-run test3
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

diogratia, 2023-12-29:
```none
%: ghdl -a test1.vhdl test2.vhdl test3.vhdl
%: ghdl -e test1
%: ghdl -e test2
%: ghdl -e test3
%: ghdl -r test1
test1.vhdl:25:5:@0ms:(report note): Arrived here.
%: ghdl -r test2
test2.vhdl:25:5:@0ms:(report note): Arrived here.
%: ghdl -r test3
test3.vhdl:24:5:@0ms:(report note): Arrived here.
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r755.g3b3c2e3ea) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```

Note the llvm 15.0.7.

kdiff3 differences:

[image]

tgingold, 2023-12-30:
Yes, it fails with mcode backend.  I am testing a fix.

commit ccb7a093 (2023-12-30): simul-vhdl_simul: keep resolution for guarded signals

Fix #2571
closed 2023-12-30 by tgingold