ghdl/ghdl #2572: raised TYPES.INTERNAL_ERROR : simul-vhdl_simul.adb:3951

smaslovski, 2023-12-30:
Hi,

Analyzing and elaborating of the code given below (fft_types.vhd, fft.vhd, and tb.vhd) goes fine, however, running the testbench results in a crash:

```
$ ghdl -a --std=08 fft_types.vhd fft.vhd tb.vhd
$ ghdl -e --std=08 tb
$ ghdl -r --std=08 tb

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r760.g224f41ed3) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/home/stas/vhdl/fft/bug/
Command line:
/home/stas/app/ghdl/bin/ghdl -r --std=08 tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : simul-vhdl_simul.adb:3951
******************************************************************
```

Here is the code:

fft_types.vhd:
```
library ieee;
use ieee.math_complex.all;
use std.textio.all;

package fft_types is
  subtype fft_data_type is complex;
  type fft_data_vector is array (natural range <>) of fft_data_type;
  pure function "+" (a, b : fft_data_vector) return fft_data_vector;
  pure function "*" (a, b : fft_data_vector) return fft_data_vector;
end package fft_types;

package body fft_types is
  pure function "+" (a, b : fft_data_vector) return fft_data_vector is
    variable res : fft_data_vector (a'range);
  begin
    for i in a'range loop
      res(i) := a(i) + b(i);
    end loop;
    return res;
  end function;

  pure function "*" (a, b : fft_data_vector) return fft_data_vector is
    variable res : fft_data_vector (a'range);
  begin
    for i in a'range loop
      res(i) := a(i) * b(i);
    end loop;
    return res;
  end function;
end package body fft_types;
```

fft.vhd:
```
library ieee;
use work.fft_types.all;

entity fft is
  generic (
    POW : natural  := 4;
    DIM : positive := 2**POW);
  port (
    x, w : in  fft_data_vector (0 to DIM-1);
    y    : out fft_data_vector (0 to DIM-1));
end entity fft;

architecture recursive of fft is
  subtype half_array is fft_data_vector (0 to DIM/2-1);
  signal even, odd, x_e, x_o, w_e : half_array;
begin
  stage :
  if base_case: POW = 0 generate
    y <= x;
  else general_case: generate
    split_even_odd :
    for i in half_array'range generate
      x_e(i) <= x(2*i);
      x_o(i) <= x(2*i+1);
      w_e(i) <= w(2*i);
    end generate;

    even_fft :
      entity fft
        generic map (
          POW => POW-1)
        port map (
          x => x_e, w => w_e, y => even);

    odd_fft :
      entity fft
        generic map (
          POW => POW-1)
        port map (
          x => x_o, w => w_e, y => odd);

    y(0 to DIM/2-1)   <= even + w(0 to DIM/2-1)*odd;
    y(DIM/2 to DIM-1) <= even + w(DIM/2 to DIM-1)*odd;
  end generate stage;
end architecture recursive;
```

tb.vhd:
```
library ieee;
use ieee.math_real.all;
use ieee.math_complex.all;
use work.fft_types.all;
use std.textio.all;

entity tb is end entity;

architecture mixed of tb is
  constant POW : natural := 2;
  constant DIM : positive := 2**POW;
  signal x, y : fft_data_vector (0 to DIM-1);

  function weight(n : positive) return fft_data_vector is
    variable w : fft_data_vector (0 to n-1);
  begin
    for k in w'range loop
      w(k) := exp(math_cbase_j*math_2_pi*real(k)/real(n));
    end loop;
    return w;
  end function;

begin

  x <= (complex'(0.0,0.0), complex'(1.0,1.0), complex'(2.0,2.0), complex'(3.0,3.0));

  fft_inst: entity work.fft(recursive)
    generic map (POW => POW)
    port map (x => x, w => weight(DIM), y => y);

  process (y) is
    variable lin : line;
    alias s is to_string [integer return string];
    alias s is to_string [real return string];
  begin
    for i in x'range loop
      lin := new string'("x(" & s(i) & ") = (" & s(x(i).re) & "," & s(x(i).im) & "), y(" & s(i) & ") = (" & s(y(i).re) & "," & s(y(i).im) & ")");
      writeline(output, lin);
      deallocate(lin);
    end loop;
  end process;

end architecture mixed;
```

BR,
Stanislav

diogratia, 2023-12-30:
```none
%: ghdl -r --std=08 tb
./tb:error: bound check failure at ../../src/ieee2008/math_complex-body.vhdl:1312
in process .tb(mixed).fft_inst@fft(recursive).stage.even_fft@fft(recursive).stage.P0
./tb:error: simulation failed
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r755.g3b3c2e3ea) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```

tgingold, 2023-12-30:
Yes, it crashes with the mcode backend but not with the llvm or gcc backend.
I am testing a fix.
However, you should initialize the signals with `(others => (0.0, 0.0))` to avoid overflows.

commit 2e6ab75a (2023-12-30): simul-vhdl_simul: handle association with a float static expr

Fix #2572
closed 2023-12-30 by tgingold