ghdl/ghdl #2575: Stat collection logic error

tfuto, 2024-01-07:
**Description**

The following code section in [grt-signals.adb#3628](https://github.com/ghdl/ghdl/blob/336786f7651eb139538cc96a9ddc5c5f6bc93507/src/grt/grt-signals.adb#L3628C8-L3628C8) first clears `Sig.Active` and `Sig.Event`, and **then** checks their value a few lines later. Consequently, `Nbr_Active` and `Nbr_Events` will never be incremented, the entire `if Options.Flag_Stats then`...`end if` confitional statement is dead code currently.

Clearing the variables should be moved after the statistics update.

```
  Sig.Active := False;
  Sig.Event := False;

  if Options.Flag_Stats then
     if Sig.Active then
        Nbr_Active := Nbr_Active + 1;
     end if;
     if Sig.Event then
        Nbr_Events := Nbr_Events + 1;
     end if;
  end if;
```

**Expected behaviour**
 - Increasing `Nbr_Active` if `Sig.Active` is true.
 - Increasing `Nbr_Events` if `Sig.Event` is true. 

**How to reproduce?**
Requires a simple code review only, but you can also put a breakpoint on line 3628 and run any VHDL design using `ghdl_mcode -r`. (I used a build with static elaboration, mcode code generator to arrive at the breakpoint).

**Context**
Commit 336786f765 (currently master).

tgingold, 2024-01-07:
Thank you for catching that.

Do you plan to do a pull request ?

tfuto, 2024-01-07:
@tgingold  Created pull request.

closed 2024-01-09 by tfuto