ghdl/ghdl #2578: Libs at \lib\ghdl\vendors are not found by default

el-coder-sb, 2024-01-10:
**Description**
If I put e.g. the osvvm lib in the `vendors` directory the osvvm lib is not detected by default. I have to pass it via the `-P` parameter.

`PS C:\GHDL_3_0\lib\ghdl\vendors> ./compile-osvvm.ps1 -v -OSVVM -Output . -Source C:\FPGA\Testbench\OSVVM\osvvm_libs\OsvvmLibraries`

**Expected behaviour**
I´m not sure if this is a bug or the intendet behaviour. To clarify this I opened this issue. I was expecting that all the vendor libs could be placed inside the `vendors` directory. 

**How to reproduce?**
`PS C:\GHDL_3_0\lib\ghdl\vendors> ./compile-osvvm.ps1 -v -OSVVM -Output . -Source C:\FPGA\Testbench\OSVVM\osvvm_libs\OsvvmLibraries`

```vhd :file: ent.vhd
library osvvm;
    use osvvm.randompkg.all;

entity use_osvvm is
end entity use_osvvm;

architecture bar of use_osvvm is
begin
end architecture bar;
```

```bat
C:\FPGA\GHDL\Tests>ghdl -a use_osvvm.vhd
use_osvvm.vhd:10:9: cannot find resource library "osvvm"
library osvvm;
        ^
...
```

Everything works fine if I put the vendor libs into the `ghdl` directory (one `.` more at Output parameter):
`PS C:\GHDL_3_0\lib\ghdl\vendors> ./compile-osvvm.ps1 -v -OSVVM -Output .. -Source C:\FPGA\Testbench\OSVVM\osvvm_libs\OsvvmLibraries`

**Context**

- OS: Win10
> ghdl --version
> GHDL 4.0.0-dev (3.0.0.r321.g472cf6443) [Dunoon edition]
> Compiled with GNAT Version: 13.1.0

tgingold, 2024-01-11:
The script compile the library, but to use it it must be on the search path (use `ghdl --disp-config` to display it).
`PREFIX\lib\ghdl` is in the search path.

el-coder-sb, 2024-01-12:
> The script compile the library, but to use it it must be on the search path (use `ghdl --disp-config` to display it). `PREFIX\lib\ghdl` is in the search path.

Thanks for that @tgingold. 

Maybe a renaming of the `/vendors` folder to `/vendors_scripts` would prevent such misinterpretation. Or changing the search to look for libs down the `/vendors` directory. Feel free to comment if one of the ideas is worth to get an issue.

As this topic itself is done I´ll close it.

closed 2024-01-12 by el-coder-sb

Paebbels, 2024-01-12:
@el-coder-sb the precompiled libraries can't be added automatically to the search path to GHDL.  
There are several reasons:
* pre-compiled libraries are usually not compiled into a system directory like `/usr/...`.
* a new pre-compiled library is needed for every new GHDL version and for every vendor tool version.

If the pre-compiled library would be automatially in search paths list, there would be no way to use my own pre-compiled library.

I made the scripts to help people precompile vendor libraries. It was hard work to figure out how to compile each individual tool, with settings, workarounds, compile-order, etc.

----------
Hint: The libraries are compiled in a structure that's directly supported by GHDL for usage with `-P`. Here you just provide the root directory of an altera or xilinx library set und GHDL resolves it to libraray name and VHDL version.

el-coder-sb, 2024-01-15:
@Paebbels Thanks for your information and many thanks for your effort for the scripts!!

I still didn´t get the whole picture. For example I can´t see why the `config.psm1` handles intel/lattice/xilinx different to osvvm/uvvm (as the default destination pattern is different)?