ghdl/ghdl #2580: VHDL-2008 generic package (with generic type) instantiation declaration in entity declaration

jeras, 2024-01-10:
Using a generic package with a type generic causes a crash. Or maybe the cause of the crash is caused by package instantiation inside the entity/architecture body. The later seems to be at least partially supported https://github.com/ghdl/ghdl/issues/1262.

I provide an example, that compiles and executes well using Questa and Rivera Pro.

```VHDL
library ieee;
use ieee.std_logic_1164.all;

package test_pkg is
generic(
    type data_type
);
    type array_data_type is array (natural range <>) of data_type;
end package test_pkg;

use std.env.all;
use std.textio.all;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test1 is
    generic (
       WIDTH : positive := 8;
       LENGTH : positive := 4
    );
    package test_pkg_def is new work.test_pkg generic map (data_type => std_logic_vector(WIDTH-1 downto 0));
    use test_pkg_def.all;
end test1;
 
architecture behavior of test1 is
begin
    testcase: process
        variable increments : array_data_type(0 to LENGTH-1);
    begin
        initloop: for i in 0 to LENGTH-1 loop
            increments(i) := std_logic_vector(to_unsigned(i, WIDTH));
        end loop initloop;
        printloop: for i in 0 to LENGTH-1 loop
            report "increments(" & integer'image(i) & ") = "  & integer'image(WIDTH) & "X" & '"' & to_hstring(increments(i)) & '"';
        end loop printloop;
        finish;
    end process testcase;
end behavior;

use std.env.all;
use std.textio.all;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test2 is
    generic (
       WIDTH : positive := 16;
       LENGTH : positive := 6
    );
end test2;
 
architecture behavior of test2 is
    package test_pkg_def is new work.test_pkg generic map (data_type => std_logic_vector(WIDTH-1 downto 0));
    use test_pkg_def.all;
begin
    testcase: process
        variable increments : array_data_type(0 to LENGTH-1);
    begin
        initloop: for i in 0 to LENGTH-1 loop
            increments(i) := std_logic_vector(to_unsigned(i, WIDTH));
        end loop initloop;
        printloop: for i in 0 to LENGTH-1 loop
            report "increments(" & integer'image(i) & ") = "  & integer'image(WIDTH) & "X" & '"' & to_hstring(increments(i)) & '"';
        end loop printloop;
        finish;
    end process testcase;
end behavior;
```

The code contains two top options:
- `test1` package instantiation inside the entity,
- `test2` package instantiation inside architecture.
Both work under Questa and Rivera Pro.

While testing on the latest GHDL git master I get:
```bash
$ ghdl -i --work=work --std=08 test.vhd 
$ ghdl -m --std=08 work.test1
$ ghdl -r --std=08 work.test1
ghdl:error: bound check failure at test.vhd:34
ghdl:error: simulation failed
$ ghdl -m --std=08 work.test2
$ ghdl -r --std=08 work.test2
ghdl:error: bound check failure at test.vhd:66
ghdl:error: simulation failed
```

I spent hours getting the example into something small and now it refuses to crash, instead I only get a bound check failure. I will probably add the larger example causing a crash later.

tgingold, 2024-01-10:
I can reproduce the error.

jeras, 2024-01-11:
I got the the crashing example to be small enough for a regression test.

~~The issue seems to be with the instance `DAT` type, which is not defined in the body of the package, but is provided to the package as a type generic. This seems to be good enough for the declaration of signals `txcDat`/`rxcDat`, but not good enough for the DUT entity instance generic. The example works if I replace the DUT instance with a simple assignment.~~

The issue does not seem to be related to generic packages at all, instead it is probably just in issue with using type generics in entity instances. The following testcase would probably be a better fit for #2458.

```VHDL
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package StreamPkg is
generic(
   type DAT
);
   type streamDataArrayType is array (natural range <>) of DAT;
end StreamPkg;

package body StreamPkg is
end package body StreamPkg;

library ieee;
use ieee.std_logic_1164.all;

entity StreamRegSlice is
   generic (
      type DAT
   );
   port (
      rxcDat_i : in  DAT;
      txcDat_o : out DAT
   );
end StreamRegSlice;

architecture rtl of StreamRegSlice is
begin
   txcDat_o <= rxcDat_i;
end rtl;

use std.env.all;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity StreamTb is
   generic (
      WIDTH : positive := 8
   );
   package StreamPkgDef is new work.StreamPkg generic map (DAT => std_logic_vector(WIDTH-1 downto 0));
   use StreamPkgDef.all;
end StreamTb;

architecture behavior of StreamTb is
-- subtype DAT is std_logic_vector(WIDTH-1 downto 0);
   signal txcDat : DAT;
   signal rxcDat : DAT;
begin

   -- DUT instance
   dut : entity work.StreamRegSlice
   generic map (
      DAT => DAT
   )
   port map (
      rxcDat_i => txcDat,
      txcDat_o => rxcDat
   );
--   rxcDat <= txcDat;

   p_test : process
   begin
      -- end simulation
      for i in 0 to 8-1 loop
         txcDat <= std_logic_vector(to_unsigned(i, WIDTH));
         wait for 10 ns;
      end loop;
      report "Simulation end";
      finish;
   end process p_test;

end behavior;
```

closed 2024-01-11 by tgingold

jeras, 2024-01-12:
The issue from the first example `bound check failure` is now fixed, thanks.

I also provided a second example causing a crash. Initially I thought it was related to generic package instantiation but it appears to be just an issue with type generics. I opened a new issue and copied a shorter example there #2584.

commit 44623143 (2024-01-11): simul-vhdl_compile: handle anonymous types for interface type.

Fix #2580