ghdl/ghdl #2584: crash related to type generics

jeras, 2024-01-12:
This was tested with a build from the latest git master.

If I replace the StreamRegSlice instantiation with just the code taken from it, the test is not crashing. So apparently this is an issue with type generics.

It might be related to #2458, which is not finished, but does not crash anymore. The main difference from my example and 2458 is that I actually have ports using the type generic.

```VHDL
library ieee;
use ieee.std_logic_1164.all;

entity StreamRegSlice is
   generic (
      type DAT
   );
   port (
      rxcDat_i : in  DAT;
      txcDat_o : out DAT
   );
end StreamRegSlice;

architecture rtl of StreamRegSlice is
begin
   txcDat_o <= rxcDat_i;
end rtl;

use std.env.all;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity StreamTb is
   generic (
      WIDTH : positive := 8
   );
end StreamTb;

architecture behavior of StreamTb is
   subtype DAT is std_logic_vector(WIDTH-1 downto 0);
   signal txcDat : DAT;
   signal rxcDat : DAT;
begin

   -- DUT instance
   dut : entity work.StreamRegSlice
   generic map (
      DAT => DAT
   )
   port map (
      rxcDat_i => txcDat,
      txcDat_o => rxcDat
   );
--   rxcDat <= txcDat;

   p_test : process
   begin
      -- end simulation
      for i in 0 to 8-1 loop
         txcDat <= std_logic_vector(to_unsigned(i, WIDTH));
         wait for 10 ns;
      end loop;
      report "Simulation end";
      finish;
   end process p_test;

end behavior;
```

closed 2024-01-12 by tgingold

jeras, 2024-01-14:
Hi, this example is still crashing, if I try to get a waveform, it is working fine without.
```VHDL
$ ghdl -r --std=08 work.StreamTb --wave=wave.ghw

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r802.g80d8aa295) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/ijeras/Cosylab/ghdl-bugreport/
Command line:
ghdl -r --std=08 work.StreamTb --wave=wave.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-rtis_utils.adb:293 invalid data
******************************************************************
```

Is there a way to run `./testsuite.sh` in `testsuite/gna/issue2584` with waveforms enabled?
Is there a manual for creating testcases.

I plan to test GHDL VHDL-2008 support a bit more in the next months.
If you would prefer, I can create future reports with a testcase as a pull request.

tgingold, 2024-01-15:
The support of 2008 features is not complete for `--wave`.  But you can use `--fst` instead (it's a different format).

commit 2567addf (2024-01-12): testsuite/gna: add a test for #2584

Close #2584