ghdl/ghdl #2587: Compilation Error (unhandled exception)

AndrzejKowalski9917, 2024-01-15:
**Description**
When compiling the minimal example provided below, the following unhandled exception is raised:

>Execution of C:\msys64\mingw64\bin\ghdl1-llvm.exe terminated by unhandled exception
raised PROGRAM_ERROR : EXCEPTION_ACCESS_VIOLATION
Load address: 0x7ff7e1060000
Call stack traceback locations:
0x7ff7e14c0bb3 0x7ffad3a5441d 0x7ffad39ce464 0x7ffad3a5340c 0x7ffa261bb7b2 0x7ffa261b9bbc 0x7ffa261b9280 0x7ffa261bd07d 0x7ffa269b3b84 0x7ffa26182b86 0x7ffa26182891 0x7ffa270e2d18 0x7ffa250150bd 0x7ffa24d379fd 0x7ff7e1063494 0x7ff7e147ff30 0x7ff7e1063266 0x7ff7e10612ec 0x7ff7e1061404 0x7ffad30f257b 0x7ffad3a0aa56
C:\msys64\mingw64\bin\ghdl.exe: compilation error

The cause is the usage of the constant `c_tmp` within the procedure when calling the stable-attribute of `s_test_signal`. If  the procedures parameter `stable_time` is given directly to the stable-attribute the case is handled with an compile error message:

> tb_playground.vhd:16:33: parameter of signal attribute must be static
    assert s_test_signal'stable(stable_time) report "Signal was not stable!" severity failure;

With an earlier version of GHDL (and propably LLVM - I reasently did an update and don't know which version was used before) I was able to circumvent the error message by using the local constant within the procedure. GHDL did not crash at compile time an the simulation ran without problems.

**Expected behaviour**
Either compile the code without errors or abort with an compilation error message.

**How to reproduce?**
The following minimal example causes the crash at compile time:

```vhd :file: tb_playground.vhd
library ieee;
use ieee.std_logic_1164.all;

entity tb_playground is
end tb_playground;

architecture sim of tb_playground is

  signal s_test_signal : std_logic := '1';

  procedure test (constant stable_time : time := 100 us) is
    constant c_tmp : time := stable_time; -- local constant to circumvent "static" error message
  begin
    assert s_test_signal'stable(stable_time) report "Signal was not stable!" severity failure;
    --                              ^ this causes the crash
  end procedure;

begin

  main: process
  begin
    wait for 10 us;
    test;
  end process;

end architecture;

```

The design is build with the following two commands, `ghdl -m` causes the crash.
```sh
ghdl -i --work=work .\tb_playground.vhd
ghdl -m --work=work tb_playground   
```
**Context**
- OS: Windows 11 (MSYS2/MINGW64)
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit SHA f686481e3`

```sh:
ghdl --version
GHDL 4.0.0-dev (3.0.0.r785.gf686481e3) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 llvm 17.0.6 code generator
Written by Tristan Gingold.
'''

tgingold, 2024-01-15:
I think this is a real bug in the VHDL LRM.  What you are trying to achieve doesn't make sense according to the LRM, but it is not forbidden by LRM rules.

I will try to investigate more.

AndrzejKowalski9917, 2024-01-15:
The NVC simulator did not like either of the two versions, so I implemented it in another way and my simulation is running again. So there is no hurry, but thanks.

tgingold, 2024-01-16:
I am testing a fix for that, but I am pretty sure the fix should be in the LRM.

nickg, 2024-01-16:
I reported a similar problem caused by constants in VHDL-2019 sequential blocks on the IEEE issue tracker but didn't get much response:

https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/281

closed 2024-01-16 by tgingold

JimLewis, 2024-08-21:
I don't think any of the signal attributes that create signals ('transaction, 'delayed, 'stable, 'quiet) are allowed to be used in a subprogram.    

That said, I would like to change that - however, I think that would require that we allow signals to be declared in sequential declarative regions - which I have seen use models for (that can be worked around if one is careful)

commit afb2a326 (2024-01-16): vhdl-sem_decls: adjust staticness of constant objects

Fix #2587