ghdl/ghdl #2589: Can no longer install pyGHDL using pip: Multiple .egg-info directories found

gralco, 2024-01-16:
**Description**
After having sucessfully installed pyGHDL (and thereby also ghdl-ls) using `pip install .` a few months back, I am no longer able to install a newer build as of today.

I have tested the following three commits, each with the same result: 2135cbf1458bd1b8b8f42bda81222ab57bd66de6 (previously installed on the Windows machine), 71287d46036e45f2bcafbc06e4b1ca1b37d48b85 (currently installed on the Fedora machine), d0693788ecfb77957d972d16ca07f27c0e76e17e (current head commit).

After seeing this issue on the Windows machine, I tested doing the `pip install .` on the Fedora machine without changing commits from what was used to install pyGHDL on it to begin with, which resulted in the same "Multiple .egg-info directories found" error.

On the Windows 10 machine I tried doing a `pip uninstall pyghdl` but that did not make it so that I am able to install a newer build. I reluctant to do the same the same on my Fedora machine out of fear of losing the existing pyGHDL install I have on it now (as what happened on Windows).

**Expected behavior**
After successfully updating GHDL via `configure --enable-synth`, `make`, `sudo make install`, I would expect that doing `pip install .` would update pyGHDL and ghdl-ls.

**How to reproduce?**
Run `pip install .`, this was done on two separate machines with different OSes that successfully installed pyGHDL using this method in the past.

Here's what I get why I attempt it (on both Fedora and Windows):
```
$ pip install . -v
Using pip 23.2.1 from /usr/lib/python3.12/site-packages/pip (python 3.12)
Defaulting to user installation because normal site-packages is not writeable
Processing /home/ekuzmenko/ghdl
  Running command pip subprocess to install build dependencies
  Collecting pyTooling>=4.0.1
    Obtaining dependency information for pyTooling>=4.0.1 from https://files.pythonhosted.org/packages/53/89/be6c6b36bb0d0fb71e915f85650f6bde40e3a27d31cd22a23ad4f4e08319/pyTooling-6.0.0-py3-none-any.whl.metadata
    Using cached pyTooling-6.0.0-py3-none-any.whl.metadata (21 kB)
  Collecting setuptools>=62.3.3
    Obtaining dependency information for setuptools>=62.3.3 from https://files.pythonhosted.org/packages/55/3a/5121b58b578a598b269537e09a316ad2a94fdd561a2c6eb75cd68578cc6b/setuptools-69.0.3-py3-none-any.whl.metadata
    Using cached setuptools-69.0.3-py3-none-any.whl.metadata (6.3 kB)
  Collecting wheel>=0.38.1
    Obtaining dependency information for wheel>=0.38.1 from https://files.pythonhosted.org/packages/c7/c3/55076fc728723ef927521abaa1955213d094933dc36d4a2008d5101e1af5/wheel-0.42.0-py3-none-any.whl.metadata
    Using cached wheel-0.42.0-py3-none-any.whl.metadata (2.2 kB)
  Using cached pyTooling-6.0.0-py3-none-any.whl (135 kB)
  Using cached setuptools-69.0.3-py3-none-any.whl (819 kB)
  Using cached wheel-0.42.0-py3-none-any.whl (65 kB)
  Installing collected packages: wheel, setuptools, pyTooling
  ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
  pyvhdlmodel 0.25.1 requires pyTooling<5.0,>=4.0.1, but you have pytooling 6.0.0 which is incompatible.
  pyghdl 4.0.0.dev0 requires pyTooling<5.0,>=4.0.1, but you have pytooling 6.0.0 which is incompatible.
  Successfully installed pyTooling-6.0.0 setuptools-69.0.3 wheel-0.42.0
  Installing build dependencies ... done
  Running command Getting requirements to build wheel
  [pyTooling.Packaging] Extracting requirements from 'pyGHDL/requirements.txt'.
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/libghdl/requirements.txt'.
  [pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/dom/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'pyGHDL/dom/../libghdl/requirements.txt'.
  [pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   - pyVHDLModel==0.25.1
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/lsp/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'pyGHDL/lsp/../libghdl/requirements.txt'.
  [pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/cli/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'pyGHDL/cli/../dom/requirements.txt'.
  [pyTooling.Packaging]       Extracting requirements from 'pyGHDL/cli/../dom/../libghdl/requirements.txt'.
  [pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     - pyVHDLModel==0.25.1
  [pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   - pyAttributes>=2.3.2
  [pyTooling.Packaging] Extracting requirements from 'doc/requirements.txt'.
  [pyTooling.Packaging]   Extracting requirements from 'doc/../pyGHDL/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/libghdl/requirements.txt'.
  [pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/dom/requirements.txt'.
  [pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/dom/../libghdl/requirements.txt'.
  [pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     - pyVHDLModel==0.25.1
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/lsp/requirements.txt'.
  [pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/lsp/../libghdl/requirements.txt'.
  [pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/cli/requirements.txt'.
[... 69 lines cut ...]
      self._bubble_up_info_directory(metadata_directory, ".egg-info")
    File "/tmp/pip-build-env-2q80wbbi/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 337, in _bubble_up_info_directory
      info_dir = self._find_info_directory(metadata_directory, suffix)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/tmp/pip-build-env-2q80wbbi/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 348, in _find_info_directory
      assert len(candidates) == 1, f"Multiple {suffix} directories found"
  AssertionError: Multiple .egg-info directories found
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /usr/bin/python3 /usr/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpwfgm4wsp
  cwd: /home/ekuzmenko/ghdl
  Preparing metadata (pyproject.toml) ... error
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

gralco, 2024-01-16:
As a sanity check, I just installed a clean Fedora 39 install and attempted to do `pip install .` within it, it had the same result.

tgingold, 2024-01-16:
There is a problem with pyTooling.  Can you uninstall it, install pyTooling version 4.0.1 and try again ?

Paebbels, 2024-01-16:
I'm right now updating pyTooling from v5.0 to v6.0 (did the release a few days ago), but this problem here would also occure with pyTooling v5.0.

Your system reports a pyTooling v6.0.0 in the cache and installs it, but all requirement rules show a pyTooling >=4.0.1, <5.0 rule. It looks like a `pip install .` doesn't obey the requirement rules.

Run this to downgrade to an exact version:
`pip install -U pyTooling==4.0.1`

or with a range:
`pip install -U pyTooling<5.0`

Paebbels, 2024-01-16:
I usually package pyGHDL before I use pip install so I might not have noticed it before.

gralco, 2024-01-16:
I tried doing `pip cache purge` and then install pyTooling version 4.0.1 but it still gets the latest version instead of obeying the <6.0 requirement:
```
[ekuzmenko@fedora ghdl]$ pip cache purge
Files removed: 9
[ekuzmenko@fedora ghdl]$ pip install -U pyTooling==4.0.1
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyTooling==4.0.1 in /home/ekuzmenko/.local/lib/python3.12/site-packages (4.0.1)
[ekuzmenko@fedora ghdl]$ pip install . -v
Using pip 23.2.1 from /usr/lib/python3.12/site-packages/pip (python 3.12)
Defaulting to user installation because normal site-packages is not writeable
Processing /home/ekuzmenko/ghdl
  Running command pip subprocess to install build dependencies
  Collecting pyTooling>=4.0.1
    Obtaining dependency information for pyTooling>=4.0.1 from https://files.pythonhosted.org/packages/53/89/be6c6b36bb0d0fb71e915f85650f6bde40e3a27d31cd22a23ad4f4e08319/pyTooling-6.0.0-py3-none-any.whl.metadata
    Downloading pyTooling-6.0.0-py3-none-any.whl.metadata (21 kB)
  Collecting setuptools>=62.3.3
    Obtaining dependency information for setuptools>=62.3.3 from https://files.pythonhosted.org/packages/55/3a/5121b58b578a598b269537e09a316ad2a94fdd561a2c6eb75cd68578cc6b/setuptools-69.0.3-py3-none-any.whl.metadata
    Downloading setuptools-69.0.3-py3-none-any.whl.metadata (6.3 kB)
  Collecting wheel>=0.38.1
    Obtaining dependency information for wheel>=0.38.1 from https://files.pythonhosted.org/packages/c7/c3/55076fc728723ef927521abaa1955213d094933dc36d4a2008d5101e1af5/wheel-0.42.0-py3-none-any.whl.metadata
    Downloading wheel-0.42.0-py3-none-any.whl.metadata (2.2 kB)
  Downloading pyTooling-6.0.0-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 136.0/136.0 kB 2.9 MB/s eta 0:00:00
  Downloading setuptools-69.0.3-py3-none-any.whl (819 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 819.5/819.5 kB 15.1 MB/s eta 0:00:00
  Downloading wheel-0.42.0-py3-none-any.whl (65 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 65.4/65.4 kB 7.5 MB/s eta 0:00:00
  Installing collected packages: wheel, setuptools, pyTooling
  ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
  pyvhdlmodel 0.25.1 requires pyTooling<5.0,>=4.0.1, but you have pytooling 6.0.0 which is incompatible.
  pyghdl 4.0.0.dev0 requires pyTooling<5.0,>=4.0.1, but you have pytooling 6.0.0 which is incompatible.
  Successfully installed pyTooling-6.0.0 setuptools-69.0.3 wheel-0.42.0
  Installing build dependencies ... done
  Running command Getting requirements to build wheel
  [pyTooling.Packaging] Extracting requirements from 'pyGHDL/requirements.txt'.
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/libghdl/requirements.txt'.
  [pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/dom/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'pyGHDL/dom/../libghdl/requirements.txt'.
  [pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   - pyVHDLModel==0.25.1
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/lsp/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'pyGHDL/lsp/../libghdl/requirements.txt'.
  [pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   Extracting requirements from 'pyGHDL/cli/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'pyGHDL/cli/../dom/requirements.txt'.
  [pyTooling.Packaging]       Extracting requirements from 'pyGHDL/cli/../dom/../libghdl/requirements.txt'.
  [pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     - pyVHDLModel==0.25.1
  [pyTooling.Packaging]   - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]   - pyAttributes>=2.3.2
  [pyTooling.Packaging] Extracting requirements from 'doc/requirements.txt'.
  [pyTooling.Packaging]   Extracting requirements from 'doc/../pyGHDL/requirements.txt'.
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/libghdl/requirements.txt'.
  [pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/dom/requirements.txt'.
  [pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/dom/../libghdl/requirements.txt'.
  [pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     - pyVHDLModel==0.25.1
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/lsp/requirements.txt'.
  [pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/lsp/../libghdl/requirements.txt'.
  [pyTooling.Packaging]       - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     Extracting requirements from 'doc/../pyGHDL/cli/requirements.txt'.
  [pyTooling.Packaging]       Extracting requirements from 'doc/../pyGHDL/cli/../dom/requirements.txt'.
  [pyTooling.Packaging]         Extracting requirements from 'doc/../pyGHDL/cli/../dom/../libghdl/requirements.txt'.
  [pyTooling.Packaging]         - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]       - pyVHDLModel==0.25.1
  [pyTooling.Packaging]     - pyTooling>=4.0.1, <5.0
  [pyTooling.Packaging]     - pyAttributes>=2.3.2
  [pyTooling.Packaging] - sphinx
[... 62 lines cut ...]
      self._bubble_up_info_directory(metadata_directory, ".egg-info")
[... cut ...]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /usr/bin/python3 /usr/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp8vhfq1_1
  cwd: /home/ekuzmenko/ghdl
  Preparing metadata (pyproject.toml) ... error
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

tgingold, 2024-01-16:
I am adding the fix.

closed 2024-01-16 by tgingold

Paebbels, 2024-01-16:
And I need to add that support before the next GHDL release (its almost Feb. 2024), right?

tgingold, 2024-01-17:
No, it's not required.
There are problems with pyGHDL in CI, I need to investigate too (or disable the packaging).

commit 5ab0b9a7 (2024-01-16): pyproject.toml: adjust PyTooling version

Fix #2589