ghdl/ghdl #2592: raised ADA.ASSERTIONS.ASSERTION_ERROR : trans.adb:442

smaslovski, 2024-01-19:
**Description**
GHDL craches when running the code given below.

**Expected behaviour**
The code works in NVC, with the following output:
```
$ nvc --std=08 -a bug2.vhd
$ nvc --std=08 -e tb
$ nvc --std=08 -r tb
y(0) = UUUUUUUU
y(1) = UUUUUUUU
y(2) = UUUUUUUU
y(3) = UUUUUUUU
y(0) = 00000001
y(1) = 11111110
y(2) = 11111111
y(3) = 00000000

**How to reproduce?**

$ ghdl -a --std=08 bug2.vhd
$ ghdl -e --std=08 tb
$ ghdl -r --std=08 tb

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r814.g0c2782bae) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/home/stas/vhdl/bug/
Command line:
/home/stas/app/ghdl/bin/ghdl -r --std=08 tb
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : trans.adb:442
******************************************************************

```vhd :file: bug2.vhd
library ieee;
use ieee.numeric_std.all;

package foo_pkg is
  constant WID : positive := 8;
  subtype data_type is signed (WID-1 downto 0);
  type data_vector is array (natural range <>) of data_type;
  function even_elem(x : data_vector) return data_vector;
  function odd_elem(x : data_vector) return data_vector;
end package;

package body foo_pkg is
  function even_elem(x : data_vector) return data_vector is
    variable res : data_vector (0 to x'length/2-1);
  begin
    for i in res'range loop
      res(i) := x(2*i);
    end loop;
    return res;
  end function;
  function odd_elem(x : data_vector) return data_vector is
    variable res : data_vector (0 to x'length/2-1);
  begin
    for i in res'range loop
      res(i) := x(2*i+1);
    end loop;
    return res;
  end function;
end package body;

library ieee;
use ieee.numeric_std.all;
use work.foo_pkg.all;

entity foo is
  generic (
    DIM : positive := 4;
    p : data_vector (0 to DIM/2-1));
  port (
    x : in  data_vector (0 to DIM-1);
    y : out data_vector (0 to DIM-1));
end entity;

architecture rec of foo is
  subtype half_array is data_vector (0 to DIM/2-1);
  signal even, odd : half_array := (others => to_signed(0, WID));
begin
  stage :
  if gen_case : DIM > 1 generate
    foo_even :
      entity foo
        generic map (DIM => DIM/2, p => even_elem(p))
        port map (x => even_elem(x), y => even);
    foo_odd :
      entity foo
        generic map (DIM => DIM/2, p => even_elem(p))
        port map (x => odd_elem(x), y => odd);
    arithm :
    process (all) is
    begin
      for i in half_array'range loop
        y(i)       <= data_type'(even(i)*p(i)) + odd(i);
        y(i+DIM/2) <= data_type'(even(i)*p(i)) - odd(i);
      end loop;
    end process;
  else base_case : generate
    y <= x;
  end generate;
end architecture;

library ieee;
use ieee.numeric_std.all;
use std.textio.all;
use work.foo_pkg.all;

entity tb is
end entity;

architecture test of tb is
  constant DIM : positive := 4;
  signal x, y : data_vector(0 to DIM-1);

  function coeff(n : positive) return data_vector is
    variable res : data_vector (0 to n-1);
  begin
    for i in res'range loop
      res(i) := to_signed(i, WID);
    end loop;
    return res;
  end function;

begin
  x <= (others => to_signed(1, WID));
  foo_inst :
    entity work.foo(rec)
      generic map (DIM => DIM, p => coeff(DIM/2))
      port map (x => x, y => y);
  log :
  postponed process (all) is
    variable lin : line;
    alias s is to_string [integer return string];
    alias s is to_string [data_type return string];
  begin
    for i in y'range loop
      write(lin, "y(" & s(i) & ") = " & s(y(i)));
      writeline(output, lin);
    end loop;
  end process;
end architecture;
```

BR,
Stanislav

smaslovski, 2024-01-20:
To add more info to this report, I tested also the following variant of the code in GHDL and modelsim:
```
architecture rec of foo is
  subtype half_array is data_vector (0 to DIM/2-1);
  signal x_e, x_o, even, odd : half_array := (others => to_signed(0, WID));
begin
  stage_n :
  if DIM > 1 generate
    x_e <= even_elem(x);
    foo_even :
      entity foo
        generic map (DIM => DIM/2, p => even_elem(p))
        port map (x => x_e, y => even);
    x_o <= odd_elem(x);
    foo_odd :
      entity foo
        generic map (DIM => DIM/2, p => even_elem(p))
        port map (x => x_o, y => odd);
    arithm :
    process (all) is
    begin
      for i in half_array'range loop
        y(i)       <= data_type'(even(i)*p(i)) + odd(i);
        y(i+DIM/2) <= data_type'(even(i)*p(i)) - odd(i);
      end loop;
    end process;
  end generate stage_n;
  stage_0 :
  if DIM = 1 generate
    y <= x;
  end generate stage_0;
end architecture;
```
In this case, modelsim produces the expected output:
```
# y(0) = UUUUUUUU
# y(1) = UUUUUUUU
# y(2) = UUUUUUUU
# y(3) = UUUUUUUU
# y(0) = 00000001
# y(1) = 11111110
# y(2) = 11111111
# y(3) = 00000000
```
On the other hand, GHDL does not crash in this case, but reports an arror (which I believe is incorrect):
```
$ ghdl -r --std=08 tb   
ghdl:error: bound check failure at bug2.vhd:64
ghdl:error: simulation failed
```

BR,
Stanislav

tgingold, 2024-01-20:
I was investigating that case.   There is a problem with:
```
 y(i)       <= data_type'(even(i)*p(i)) + odd(i);
```
Both `even` and `p` are vector of length `WID`, so they cannot be type qualified to `data_type` (or assigned to `y`.
You need to resize the result of the multiplication as it is too wide.
Probably modelsim is skipping a bound check.

smaslovski, 2024-01-20:
Hi,

Both even(i) and p(i) are of type data_type. The width of data_type is WID bits. The product of even(i) and p(i) would normaly require 2*WID bits of storage. But, as far as I undertand, qualifying the multiplication result type to data_type should allow truncation of the result to WID bits. I think NVC and modelsim do just that in this case. However, I have to read through the standard to confirm if that is a correct semantics.

BR,
Stanislav

tgingold, 2024-01-20:
No, qualifying a value do not result in a truncation AFAIU.  A quote would be very interesting!

commit 7cbb8e9b (2024-01-20): trans-chap9: do not generate association functions

Fix #2592 (for mcode)
closed 2024-01-20 by tgingold