ghdl/ghdl #2595: GHDL mcode package dropped from MSYS2 MINGW32 on Dec 21, 2023

ugrossm, 2024-01-23:
Hi there,
I just realised that I cannot update GHDL mcode through MSYS2 pacman on Windows anymore.
See commit  https://github.com/msys2/MINGW-packages/commit/267c3ca8dcffb39b5c39d8bd6031229268e485bf
I opened an issue there https://github.com/msys2/MINGW-packages/issues/19843 asking to re-enable the package.
Any news regarding 64bit support for mcode?

Regards, 
Uwe

tgingold, 2024-01-23:
I plan to add it this year.
Currently, it somewhat already works.  Only on case of failure (assertion, finish, bound error), it behaves incorrectly.
(mcode needs to generate unwind info for win64).

pidgeon777, 2024-02-11:
> I plan to add it this year. Currently, it somewhat already works. Only on case of failure (assertion, finish, bound error), it behaves incorrectly. (mcode needs to generate unwind info for win64).

Hello, is it the same problem reported here?:

https://github.com/ghdl/ghdl/issues/2556

tgingold, 2024-02-11:
Yes, same root cause.

pidgeon777, 2024-02-11:
What a quick reply, thank you. Hopefully this will be solved 🤞.

tgingold, 2024-04-10:
So mcode is now supported on windows 64.

closed 2024-04-10 by tgingold

Paebbels, 2024-04-10:
@tgingold do we need to review the documentation for that big change?

I think there have been overviews what environment to use for what usecase and hints like MinGW64 mcode doesn't support simulation yet. Or have you already scanned the documentation for that change?

What about build instructions etc. do we maintain them further or do we mark them as deprecated for MinGW32?

tgingold, 2024-04-11:
Yes, we need to review the documentation.

I don't think build instructions need to be changed, as that's the same as MinGW32 (which we are currently keeping).