ghdl/ghdl #2597: Error: symbol `work__test1__ARCH__rtl__p_proc__OT__STL' is already defined

avelure, 2024-01-26:
**Description**
There is an issue with external names of vector signals combined with for generate loops with the GCC backend when analysing. 
std_logic works ok.
LLVM/MCODE is ok.

**Expected behaviour**
No error.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture rtl of test is
  signal sig  : std_logic_vector(1 downto 0);
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test1 is
end entity;

architecture rtl of test1 is
begin
  g_test : for i in 0 to 1 generate
  begin
    i_test : entity work.test;
  end generate;
  p_proc : process
    alias sig    is <<signal g_test(0).i_test.sig : std_logic_vector(1 downto 0)>>;
    alias sig1   is <<signal g_test(1).i_test.sig : std_logic_vector(1 downto 0)>>;
  begin
    wait;
  end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-gcc
ghdl -a --std=08 ent.vhd
```
```
test.s: Assembler messages:
test.s:447: Error: symbol `work__test1__ARCH__rtl__p_proc__OT__STL' is already defined
ghdl:error: compilation error
```

**Context**

- OS: linux
- Origin:  nightly 9e2b6fc95

tgingold, 2024-01-27:
Ok, but I am not sure what should I do as external names are not supported by the GCC backend.
So even if it can be assembled, it would fail later.

avelure, 2024-01-27:
Sure, I'm only expecting analysis to pass, I use mcode for this test. I won't simulate this testbench file with gcc, but it is compiled with all the others in my script. Sometimes I would like to use the gcc backend for one of the testbenches or to just do linting with all the backends to check for ghdl bugs early. I would prefer not to have exceptions in the scripts for each backend.

commit b8fb99b0 (2024-01-28): trans-chap4: prefix subtype of external name

Fix #2597
closed 2024-01-28 by tgingold