ghdl/ghdl #2598: Sensitivity warnings

avelure, 2024-01-26:
**Description**
I enabled `-Wsensitivity` on a larger project and it seems there are several cases that are not handled correctly.

**Expected behaviour**
No warnings, these should not require that the signal be added to the sensitivity list as they are masked behind another event.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture rtl of test is
 type t_record is record
   test : std_logic;
 end record;
 signal test1 : t_record;
 signal test2 : std_logic_vector(1 downto 0);
 signal test3_0 : std_logic;
 signal test3_1 : std_logic;
 signal test3_2 : natural;
 signal test3_3 : natural;
 signal test3_4 : std_logic;
 signal test3_5 : time;
 signal test4_0 : std_logic;
 signal test4_1 : std_logic;
begin
  p_proc_1 : process(test1.test)
  begin
  if rising_edge(test1.test) then
    null;
  end if;
  end process;

  p_proc_2 : process(test2(1))
  begin
  if rising_edge(test2(1)) then
    null;
  end if;
  end process;

  p_proc_3 : process(test3_0, test3_1, test3_4)
    variable v_test3_2 : natural;
  begin
    if test3_0 then
      if rising_edge(test3_1) then
        if test3_2 /= v_test3_2 then
          v_test3_2 := v_test3_2 + 1;
        end if;
        if test3_5 < 0 ns then
          null;
        end if;
        test3_3 <= test3_3 + 1;
      end if;
    elsif test3_4'event then
      null;
    end if;
  end process;

  p_proc_4_0 : process(test4_0, test4_1)
    variable v_test3_2 : natural;
  begin
    if rising_edge(test4_0) then
      null;
    elsif test4_1'event then
      null;
    end if;
  end process;

  p_proc_4_1 : process(test4_0, test4_1)
    variable v_test3_2 : natural;
  begin
    if test4_1'event then
      null;
    elsif rising_edge(test4_0) then
      null;
    end if;
  end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 -Wsensitivity -Werror ent.vhd
```

```
test_sensitivity.vhd:25:24: incomplete sensitivity list, signal "test1" is missing
  if rising_edge(test1.test) then
                       ^
test_sensitivity.vhd:23:22: extra signal "test1" in sensitivity list
  p_proc_1 : process(test1.test)
                     ^
test_sensitivity.vhd:32:23: incomplete sensitivity list, signal "test2" is missing
  if rising_edge(test2(1)) then
                      ^
test_sensitivity.vhd:30:22: extra signal "test2" in sensitivity list
  p_proc_2 : process(test2(1))
                     ^
test_sensitivity.vhd:42:12: incomplete sensitivity list, signal "test3_2" is missing
        if test3_2 /= v_test3_2 then
           ^
test_sensitivity.vhd:45:12: incomplete sensitivity list, signal "test3_5" is missing
        if test3_5 < 0 ns then
           ^
test_sensitivity.vhd:48:20: incomplete sensitivity list, signal "test3_3" is missing
        test3_3 <= test3_3 + 1;
                   ^
test_sensitivity.vhd:55:33: extra signal "test4_1" in sensitivity list
  p_proc_4_0 : process(test4_0, test4_1)
                                ^
test_sensitivity.vhd:65:24: extra signal "test4_0" in sensitivity list
  p_proc_4_1 : process(test4_0, test4_1)
```

**Context**

- OS: win11
- Origin: from source mingw32 179e8e8c1

tgingold, 2024-01-27:
Yes, this is as documented.
This warning cannot be perfect.  In particular `proc_3` and `proc_4` are not supposed to be synthesized so they are outside the scope of this warning.
I will see what I can do for `proc_1` and `proc_2`.

tgingold, 2024-01-28:
I have fixed the `proc_1` case.
I think `proc_2` could be handled the same way, except the index may not be known at analysis time.

I am closing this issue, maybe it could be reopen if the array case is common.

closed 2024-01-28 by tgingold

avelure, 2024-01-29:
Thanks, it was only proc_1 that was in real rtl, proc_2 was in verification code embedded in the RTL, while the others were in testbench code.