ghdl/ghdl #2599: SYSTEM.ASSERTIONS.ASSERT_FAILURE : elab-vhdl_objtypes.adb:1183

JulianKemmerer, 2024-01-28:
**Description**
Attempting to use GHDL plugin for yosys to synthesize a simple blinking LED design. 

**Expected behaviour**
Normal behavior, completing yosys syn - have old version
```
GHDL 3.0.0-dev (2.0.0.r1094.gb15423880.dirty) [Dunoon edition]
```
which does not show the issue :shrug: 

**How to reproduce?**
Unzip the attached file.
[pipelinec_output_blink.c_159443.zip](https://github.com/ghdl/ghdl/files/14074461/pipelinec_output_blink.c_159443.zip)

```
cd examples/blink.c/blink
bash blink_0CLK_a5a1cd4e_top.sh 
```
`.sh` runs `yosys -m ghdl -p` command line pointing at a handful of included VHDL files.

**Context**
`OSS CAD Suite Linux` as of 1/27/24
`Yosys 0.37+34 (git sha1 80511ced7, clang 14.0.0-1ubuntu1.1 -fPIC -Os)`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r838.g499cda10c.dirty) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/home/julian/Desktop/pipelinec_output_blink.c_159443/examples/blink.c/blink/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : elab-vhdl_objtypes.adb:1183
******************************************************************
```

Thanks for your time!

tgingold, 2024-01-28:
The `vhdl_files.txt` contains `top.vhd` which is not in the archive.  Is it expect ?

JulianKemmerer, 2024-01-28:
Apologies for any confusion.
In this case I don't believe it is used. There is a top file with a hash extension appended to name in the reproducing script that should be used.

closed 2024-01-28 by tgingold

JulianKemmerer, 2024-01-28:
Thank you! :muscle:

commit a1e95654 (2024-01-28): vhdl-sem_expr: propagate implicit type conversion

Fix #2599