ghdl/ghdl #2600: GHDL reports `external name and object have different type` when types are the same

TrAyZeN, 2024-02-06:
**Description**
GHDL reports `external name and object have different type` when using an external name of a type, equivalent to the external type, defined in the current library.

**Expected behaviour**
I don't know what the standard says about this behaviour but it compiles and runs under modelsim. I would expect GHDL to successfully compile and run.

**How to reproduce?**
```vhd :file: rom.vhdl
library ieee;
    use ieee.std_logic_1164.all;

entity rom is
end entity rom;

architecture rtl of rom is
    type mem_t is array (31 downto 0) of std_logic_vector(31 downto 0);

    signal mem : mem_t;
begin
end architecture;
```

```vhd :file: rom_tb.vhdl
library ieee;
    use ieee.std_logic_1164.all;

library librom;
    use librom.rom;

entity rom_tb is
end entity;

architecture rtl of rom_tb is
begin
    dut : entity librom.rom;

    proc : process is
        type mem_t is array (31 downto 0) of std_logic_vector(31 downto 0);

        alias mem is <<signal .rom_tb.dut.mem : mem_t>>;
    begin
        wait for 5 ms;
    end process;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 --work=librom rom.vhdl
ghdl -a --std=08 --work=librom_test rom_tb.vhdl
ghdl -e --std=08 --work=librom_test rom_tb
```

**Context**
- OS: Linux 6.6.10
- Origin: Nightly release 9e2b6fc95

tgingold, 2024-02-06:
No, ghdl is correct.  These are two different types (`type` appears twice!).

You need to declare the type in a package, and use the package both in `rom` and in `tb_rom`.

Sorry, that's strong typing.

Paebbels, 2024-02-06:
>  but it compiles and runs under modelsim.

I don't think ModelSim has such a bug, but if so, please report it to Siemens EDA, because it would be a severe bug :) !

diogratia, 2024-02-06:
And if Modelsim doesn't catch the error the fault would likely be from the same root cause for the lack of catching this error in the llvm  and presumably gcc version of ghdl.  Incremental features being added for a new revision of the standard, where significant code restructuring can be required here noting that the error message originates from ghdl/src/synth/elab-vhdl_expr.adb:230 and the mcode version and llvm/gcc versions don't share code generation.

A search finds that Modelsim was fixing external names at least as late as 10.3a including one evaluating an external name subtype indication. A vsim error code for this bug doesn't show up in a 2011 dump of verror -all and I don't have access to a current Modelsim. That brings up the question of by what version of Modelsim the error is not detected.

The proper solution to determining whether or not the error is valid would be to reference the LRM (here IEEE Std 1076-2008):

14.4.2.3 Type declarations  

>Elaboration of a type declaration generally consists of the elaboration of the definition of the type and the
creation of that type. For a constrained type declaration that declares a partially or fully constrained
composite subtype, however, elaboration consists of the elaboration of the equivalent anonymous
unconstrained type followed by the elaboration of the named subtype of that unconstrained type.

With two separate type declarations for a type named mem_t, one in the block declarative region of the architecture of rom and the second in the process proc of the architecture of rom_tb you have created two separate types.

8.7 External names

>It is an error when evaluating an external name if the identified declarative region does not contain a
declaration of an object whose simple name is the object simple name of the external pathname. It is also an
error when evaluating an external name if the object denoted by an external constant name is not a constant,
or if the object denoted by an external signal name is not a signal, or if the object denoted by an external
variable name is not a variable. Moreover, it is an error if the base type of the object denoted by an external
name is not the same as the base type of the type mark in the subtype indication of the external name.

The two base types are from two different declarations. The last sentence specifies that is an error.

You also see that the error is detected during evaluation of the external name in the mcode version of ghdl but unfortunately not the llvm version (and likely not the gcc version).

```none
%: $GHDL -e --std=08 --work=librom rom_tb
rom_tb.vhdl:17:43:error: external name and object have different type
        alias mem is <<signal .rom_tb.dut.mem : mem_t>>;
                                          ^
%: $GHDL --version
GHDL 4.0.0-dev (3.0.0.r822.g179e8e8c1) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 static elaboration, mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```
The code for rom.vhdl and rom_tb.vhdl would make a great test for an eventual fix for the ghdl llvm/gcc versions with the expectation an error should occur. The lack of an error appears specific to evaluating an external name in this case.

TrAyZeN, 2024-02-07:
Thank you for clarifying my issue. I thought types had structural equivalence in VHDL based on my experience with ModelSim. I will check the behaviour in recent versions of ModelSim when I get time.

closed 2024-02-07 by TrAyZeN