ghdl/ghdl #2602: Crash due to shadowing

cderrien, 2024-02-06:
A ghdl bug occurs if an input file contains a procedure which has a parameter which shadows a signal.
If the file is not analysed in vhdl 2008 mode, ghdl does not crash.
The signal needs to be a vector. If the signal's type is an array type that was explicitly declared, it also does not result in a crash.

```vhd shadow.vhdl
entity shadow is
end entity;

architecture behaviour of shadow is

  signal s : bit_vector(3 downto 0);

begin

  process

    procedure example (s : in bit_vector) is
    begin
      s <= s;
    end procedure;

  begin
    wait;
  end process;

end architecture;
```

```sh
ghdl -a --std=08 -Wall shadow.vhdl
```

Ghdl was compiled from source (commit https://github.com/ghdl/ghdl/commit/f9086ac6c678aaf380493632c89254b4c3d81d8a), with the llvm back-end, on Ubuntu 22.
See bug box and the error message below.

```
is_object_name_fully_constrained: cannot handle IIR_KIND_INTERFACE_CONSTANT_DECLARATION (shadow.vhdl:13:24)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r857.gf9086ac6c) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
<redacted>
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -Wall -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -fpic -c -o shadow.o shadow.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x563812ed70f4 0x563812ed3df7 0x563813084817 0x563813087629 0x563813087761 0x56381304f06f 0x563813061944 0x563813061b9b 0x563813087741 0x563813088544 0x563813088d33 0x563813088f8c 0x5638130890f9 0x56381304a729 0x563813052077 0x563813024382 0x56381314d258 0x56381314f031 0x563813150685 0x563812d02ff1 0x7fbac7caad8e 0x7fbac7caae3e 0x563812d01823 0xfffffffffffffffe
******************************************************************
```

tgingold, 2024-02-06:
I have a fix for this (but it's not due to shadowing).

tgingold, 2024-02-09:
Now fixed (bad issue ref in message for #1602).

closed 2024-02-09 by tgingold