ghdl/ghdl #2605: GRLIB Synthesis crash when synthesizing for multiple cpu cores

JJ-Gaisler, 2024-02-11:
**Description**
GHDL synth crashes when trying to synthesize the NOEL-V cpu with more cores than one.

**How to reproduce?**

1. Download [grlib](https://www.gaisler.com/products/grlib/grlib-gpl-2023.4-b4288.tar.gz)
2. cd to grlib-gpl-2023.4-b4288/designs/noelv-generic
3. Edit config.vhd and set CFG_NCPU := (2) (anything greater than 1 crashes)
4. make ghdl
5. Run the following synthesis command

```
 ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments \
 --warn-no-binding --workdir=gnu/work --work=work \
 -Pgnu/grlib -Pgnu/synplify -Pgnu/techmap -Pgnu/wizl -Pgnu/eth \
 -Pgnu/opencores -Pgnu/gaisler -Pgnu/work \
 work.noelvmp
 ```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `45fcff5bcdebcf250a4d767f4f29062c8ceeac32`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r871.g45fcff5bc) [Dunoon edition]
Compiled with GNAT Version: Community 2021 (20210519-103)
Target: x86_64-pc-linux-gnu
/tmp/ghdl-test/grlib-gpl-2023.4-b4288/designs/noelv-generic/
Command line:
/gsl/tools/ghdl/bin/ghdl --synth --out=verilog -fexplicit --ieee=synopsys --mb-comments --warn-no-binding --workdir=gnu/work --work=work -Pgnu -Pgnu/grlib -Pgnu/synplify -Pgnu/techmap -Pgnu/wizl -Pgnu/opencores -Pgnu/gaisler -Pgnu/eth -Pgnu/work noelvmp
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : synth-vhdl_decls.adb:70
Call stack traceback locations:
0x681ff0 0x5aae2e 0x5abec5 0x5ab9ee 0x5eee18 0x5ef436 0x5ef5e1 0x5eb77d 0x5ef889 0x5eb5e5 0x5f043e 0x5f08b8 0x5d8a20 0x5dc9e1 0x5b29c0 0x65857a 0x6350d9 0x409fd4 0x7ffff7c9324b 0x40a028 0xfffffffffffffffe
******************************************************************
[Inferior 1 (process 444) exited with code 02]

```

Stack-trace 
```
Breakpoint 1, <__gnat_raise_exception> (e=0xa2d1a0 <ada.assertions.assertion_error>, message=...) at a-except.adb:969
969     a-except.adb: No such file or directory.
@(gdb) bt
#0  <__gnat_raise_exception> (e=0xa2d1a0 <ada.assertions.assertion_error>, message=...) at a-except.adb:969
#1  0x0000000000681ff2 in system.assertions.raise_assert_failure (msg=...) at s-assert.adb:46
#2  0x00000000005aae30 in synth.vhdl_decls.create_var_wire (syn_inst=syn_inst@entry=0x12ce290, decl=decl@entry=183024, kind=kind@entry=wire_variable,
    init=...) at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_decls.adb:70
#3  0x00000000005abec7 in synth.vhdl_decls.synth_variable_declaration (is_subprg=true, decl=183024, syn_inst=0x12ce290)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_decls.adb:532
#4  synth.vhdl_decls.synth_declaration (syn_inst=syn_inst@entry=0x12ce290, decl=decl@entry=183024, is_subprg=is_subprg@entry=true,
    last_type=last_type@entry=0) at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_decls.adb:729
#5  0x00000000005ab9f0 in synth.vhdl_decls.synth_declarations (syn_inst=0x12ce290, decls=<optimized out>, is_subprg=is_subprg@entry=true)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_decls.adb:857
#6  0x00000000005eee1a in synth.vhdl_stmts.synth_dynamic_subprogram_call (init=..., call=19718800, sub_inst=0x12ce290, syn_inst=0x12cbc90)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:2838
#7  synth.vhdl_stmts.synth_subprogram_call (syn_inst=syn_inst@entry=0x12cbc90, call=call@entry=239863, init=...)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:3105
#8  0x00000000005ef438 in synth.vhdl_stmts.synth_subprogram_call (syn_inst=syn_inst@entry=0x12cbc90, call=call@entry=239863)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:3152
#9  0x00000000005ef5e3 in synth.vhdl_stmts.synth_procedure_call (syn_inst=0x12cbc90, stmt=stmt@entry=239862)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:3210
#10 0x00000000005eb77f in synth.vhdl_stmts.synth_sequential_statement (c=..., stmt=239862)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4138
#11 0x00000000005ef88b in synth.vhdl_stmts.synth_sequential_statements (stmts=<optimized out>, c=...)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4200
#12 synth.vhdl_stmts.synth_if_statement (c=..., stmt=stmt@entry=239796) at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:1113
#13 0x00000000005eb5e7 in synth.vhdl_stmts.synth_sequential_statement (c=..., stmt=239796)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4099
#14 0x00000000005f0440 in synth.vhdl_stmts.synth_sequential_statements (stmts=<optimized out>, c=...)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4200
#15 synth.vhdl_stmts.synth_process_statement (proc=238686, syn_inst=0x0) at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4375
#16 synth.vhdl_stmts.synth_concurrent_statement (syn_inst=syn_inst@entry=0x10cd740, stmt=stmt@entry=238686)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4898
#17 0x00000000005f08ba in synth.vhdl_stmts.synth_concurrent_statements (syn_inst=syn_inst@entry=0x10cd740, stmts=<optimized out>)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_stmts.adb:4964
#18 0x00000000005d8a22 in synth.vhdl_insts.synth_instance_design (syn_inst=syn_inst@entry=0x10cd740, entity=entity@entry=154434, arch=arch@entry=199314)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_insts.adb:1720
#19 0x00000000005dc9e3 in synth.vhdl_insts.synth_instance (inst=...) at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_insts.adb:1805
#20 synth.vhdl_insts.synth_all_instances () at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synth-vhdl_insts.adb:1823
#21 0x00000000005b29c2 in synthesis.synth_design (design=design@entry=58614, inst=inst@entry=0xf66dd0, encoding=<optimized out>)
    at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/synthesis.adb:77
#22 0x000000000065857c in ghdlsynth.perform_action (cmd=..., args=...) at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdlsynth.adb:598
#23 0x00000000006350db in ghdlmain.main () at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdlmain.adb:672
#24 0x000000000065cec7 in ghdl_gcc () at /gsl/tools/ghdl-tmp/ghdl_sources/gcc-12.2.0/gcc/vhdl/ghdldrv/ghdl_gcc.adb:43
#25 0x0000000000409fd6 in main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>)
    at /gsl/tools/ghdl-tmp/ghdl_sources/build/gcc-objs/gcc/b~ghdl_gcc.adb:976
#26 0x00007ffff7c9324d in __libc_start_main () from /lib64/libc.so.6
#27 0x000000000040a02a in _start () at ../sysdeps/x86_64/start.S:120

```

tgingold, 2024-02-12:
I am working on the fix.

closed 2024-02-14 by tgingold

JJ-Gaisler, 2024-02-15:
Thank you for your work, I just tested the fixes and it seems to work as intended! :)

tgingold, 2024-02-15:
You're welcome.

commit 99f528ee (2024-02-14): synth-vhdl_insts: finalize value_const of packages.

Fix #2605