ghdl/ghdl #2607: External name in component instance causes "exec_pathname_concurrent_region: cannot handle IIR_KIND_COMPONENT_DECLARATION"

nickg, 2024-02-13:
**Description**
If an external name has a relative pathname that crosses a component instance, GHDL crashes with the above message.

**Expected behaviour**
The test case below should pass.

**How to reproduce?**

```vhd :file: ent.vhd
entity sub1 is
end entity;

architecture test of sub1 is
    signal t : natural;
begin

    p1: process is
        alias s is <<signal ^.s : natural>>;
    begin
        s <= 5;
        wait for 0 ns;
        assert t = 42;
        wait;
    end process;

end architecture;

-------------------------------------------------------------------------------

entity sub2 is
end entity;

architecture test of sub2 is
begin

    p2: process is
        alias t is <<signal ^.u1.t : natural>>;
    begin
        t <= 42;
        wait;
    end process;

end architecture;

-------------------------------------------------------------------------------

entity ent is
end entity;

architecture test of ent is
    signal s : natural;

    component sub1 is
    end component;

    component sub2 is
    end component;
begin

    u1: component sub1;
    u2: component sub2;

    p3: process is
    begin
        assert s = 0;
        wait for 0 ns;
        assert s = 5;
        wait;
    end process;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 7400f2449298

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
exec_pathname_concurrent_region: cannot handle IIR_KIND_COMPONENT_DECLARATION (ename7.vhd:44:15)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r875.g7400f2449298) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/tmp/
Command line:
ghdl -e --std=08 ename7
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

diogratia, 2024-02-14:
The issue appears peculiar to the mcode version:

```none
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r875.g7400f2449) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: ghdl -a --std=08 ent.vhdl
%: ghdl --elab-run --std=08 ent
%: 
```

tgingold, 2024-02-14:
Yes, but it's probably because external names are not handled by gcc or llvm backends.

commit 2cfbf41d (2024-02-15): exec_pathname_concurrent_region: handle implicit block for component

Fix #2607
closed 2024-02-15 by tgingold