ghdl/ghdl #2608: ghdl-formal specific keep attribute

Topi-ab, 2024-02-16:
**Is your feature request related to a problem? Please describe.**
Using `keep` attribute helps to debug formal verification results. One can add `dbg_this_and_that` signal which describes the state of the DUT. But to see that on formal waveform, `keep` attribute needs to be added.
Now ths industry standard `keep` attribute is also obeyed by synthesis software, and these non-needed (on production) nets/cells are kept. 

**Describe the solution you'd like**
Preferably an attribute, like `keep_formal`, which works like keep attribute, but only when formal synthesis is active (or command line option enabling it).

**Describe alternatives you've considered**
Manually commenting / uncommenting the attributes.

**Additional context**
-

tgingold, 2024-02-17:
I am not sure adding a ghdl specific attribute would be a good idea.  What about using the `keep` attribute like this:

```vhdl
package formal_keep_pkg is
  constant formal_keep : string := "true";
end;

...
  attribute keep of s : signal is formal_keep;
...
```

Topi-ab, 2024-02-17:
An old method is better than a bag of new ones... Tnx

closed 2024-02-21 by tgingold