ghdl/ghdl #2610: reopen #548 ("external names support")

mfro0, 2024-02-19:
**Description**
external names support was closed as completed in issue #548 and indeed, external names appear to work fine with the mcode backend. However, they fail with `error: NULL access dereferenced` with the llvm backend.

**Expected behaviour**
external names dereferencing should work in the llvm backend as it does with mcode.

**How to reproduce?**
the provided test case in #548 just needs to be expanded with a single report statement (within the `proc` process) that references the external name:

`report "tb_sig = " & to_string(tb_sig) severity note;`

which will just produce an `error: NULL access dereferenced` error with the llvm backend (while, as said, work fine with the mcode backend).

**Context**

- OS:
`Linux xxx 6.5.0-18-generic #18~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb  7 11:40:03 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `2b9739682` (latest from this morning)

tgingold, 2024-02-19:
Yes, external names are only supported with the mcode backend.

nathanaelhuffman, 2024-02-22:
Is there documentation of this anywhere, and is the final, expected state of things? 

I just bumped into this today and spent a bit of time trying to figure out what happened.  In my case (GHDL 4.0.0-dev (3.0.0.r823.g9e2b6fc95.dirty) [Dunoon edition]
 Compiled with GNAT Version: 10.5.0
 llvm 14.0.0 code generator), I'm seeing just an exit 255 with no stdout or stderr when trying to access a signal with an external name and since I'm running with vunit I needed to go digging a bit to figure out what was going on.

This compiles fine and runs until something attempts to access the signal and this feels somewhat unintuitive, either getting a compile error saying they're un-supported or better runtime error would be more helpful.

tgingold, 2024-02-25:
No, the final expected state of things is the support of external names with the llvm backend.