ghdl/ghdl #2611: preserving instance/signal/... letter case in waveform dump files FST/GHW

jeras, 2024-02-19:
Would it be possible to preserve the letter case in waveform dump files (end perhaps error/warning reports).

We are using [CamelCase](https://en.wikipedia.org/wiki/Camel_case) in our source files and the names in the waveform viewer become difficult to distinguish.

Since VHDL is not case sensitive, it is possible to use different letter casing for the same signal in the design, but each type, signal, variable, instance, ... has a single definition, which could be used in waveform dumps.

closed 2024-02-21 by tgingold

jeras, 2024-02-21:
This was fast, I expected a discussion about how much work this would be, but I am not complaining.

Did you also update other formats (VSD, GHW)?

Another case, where source letter case could be used are warning/error messages. As I remember (I did not re-check right now) Vivado is using lowercase and Questa is using source case.

jeras, 2024-02-21:
I tested it, it works, thanks!

tgingold, 2024-02-26:
For VCD and GHW formats, I plan to rewrite those dumpers (but not immediately), so it will come.

I have changed the errors formatting so that case is preserved.  There might be some missing messages, please create a new issue when you find one of them.

commit de79cbe6 (2024-02-21): simul-fst: use name from sources (keep case)

Fix #2611