ghdl/ghdl #2612: build on rasbian failing

evanfoss, 2024-02-23:
**Description**
Building GHDL fails on raspberry pi (P400) running rasbian. It looks like some segment of the build is redirected to x86 architecture where it predictably fails. This is a segment of the end of the make phase. Thank you for your time.

[
[ghdl-config.txt](https://github.com/ghdl/ghdl/files/14381933/ghdl-config.txt)
[ghdl-make.txt](https://github.com/ghdl/ghdl/files/14381934/ghdl-make.txt)
](url)

aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/ortho/mcode/elf_common.adb
aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-disps.adb
aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb
aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-insns.adb
aarch64-linux-gnu-gcc-10 -c -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A ortho_code-x86-flags.ads
aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-opts.adb
aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/synth/synth-ieee-utils.adb
aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/ortho/mcode/elf_arch32.ads
aarch64-linux-gnu-gcc-10 -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -I./src/simul -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -gnatw.A -I- /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-flags_linux.ads
aarch64-linux-gnu-gnatbind-10 -aI./src -aI./src/vhdl -aI./src/verilog -aI./src/synth -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/mcode -aI./src/synth -aI./src/simul -static -x ghdl_jit.ali
aarch64-linux-gnu-gnatlink-10 ghdl_jit.ali -o ghdl_mcode -g memsegs_c.o jumps.o times.o grt-cstdio.o grt-cgnatrts.o grt-no_sundials_c.o grt-cvpi.o grt-cvhpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o -ldl -lm -Wl,--version-script=/home/evan/Documents/build/ghdl/./src/grt/grt.ver -Wl,--export-dynamic -shared-libgcc
/usr/bin/ld: ./ortho_code-x86-abi.o: in function `ortho_code__x86__abi__link_intrinsics':
/home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:876: undefined reference to `__muldi3'
/usr/bin/ld: /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:876: undefined reference to `__muldi3'
/usr/bin/ld: /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:880: undefined reference to `__divdi3'
/usr/bin/ld: /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:880: undefined reference to `__divdi3'
/usr/bin/ld: /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:884: undefined reference to `__moddi3'
/usr/bin/ld: /home/evan/Documents/build/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:884: undefined reference to `__moddi3'
collect2: error: ld returned 1 exit status
aarch64-linux-gnu-gnatlink-10: error when calling /usr/bin/aarch64-linux-gnu-gcc-10
gnatmake: *** link failed.
make: *** [Makefile:217: ghdl_mcode] Error 4

**Expected behaviour**
completed build

**How to reproduce?**
./configure --prefix=/home/username/opt/ghdl
make

- OS: rasbian

tgingold, 2024-02-23:
Yes, you have to build with llvm backend (configure using `--with-llvm-config`)

evanfoss, 2024-02-24:
> Yes, you have to build with llvm backend (configure using `--with-llvm-config`)

Thanks but that faults differently. 

config and from the end of make...

./configure --with-llvm-config --prefix=/home/evan/opt/
gcc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Use full IEEE library
Build machine is: aarch64-linux-gnu
Debugging is enabled with llvm 11.0.1
Creating Makefile
Creating default_paths.ads
Creating ghdl.gpr
Creating scripts/gcc/Make-lang.in
for d in ieee/v87 ieee/v93 ieee/v08 std/v87 std/v93 std/v08 src/ieee src/ieee/v87 src/ieee/v93 src/ieee2008 src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 src/upf src/vital95 src/vital2000; do \
  mkdir -p lib/ghdl/$d; \
done
Generate ghdlsynth_maybe.ads
Generate grt-readline.ads

-------------

aarch64-linux-gnu-gcc-10 -c -I./ -I./src/ortho/llvm6 -I./src/ortho -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm6 -I./src/synth -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -I- /home/evan/Documents/build/ghdl/src/verilog/verilog-bn_tables.ads
aarch64-linux-gnu-gcc-10 -c -I./ -I./src/ortho/llvm6 -I./src/ortho -I./src -I./src/vhdl -I./src/verilog -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm6 -I./src/synth -gnat12 -gnaty3befhkmr -g -gnatwa -gnatwC -gnatf -gnata -I- /home/evan/Documents/build/ghdl/src/verilog/verilog-macros.ads
aarch64-linux-gnu-gnatbind-10 -aI./src/ortho/llvm6 -aI./src/ortho -aI./src -aI./src/vhdl -aI./src/verilog -aI./src/synth -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm6 -aI./src/synth -E -x ortho_code_main.ali
aarch64-linux-gnu-gnatlink-10 ortho_code_main.ali -o ghdl1-llvm -g llvm-cbindings.o grt-cstdio.o grt-cdynload.o --LINK=clang++ -L/usr/lib/llvm-11/lib -lLLVM-11
/usr/bin/ld: grt-cdynload.o: undefined reference to symbol 'dlsym@@GLIBC_2.17'
/usr/bin/ld: /lib/aarch64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
clang: error: linker command failed with exit code 1 (use -v to see invocation)
aarch64-linux-gnu-gnatlink-10: error when calling /usr/bin/clang++
gnatmake: *** link failed.
make[1]: *** [src/ortho/llvm6/Makefile:14: ghdl1-llvm] Error 4
make[1]: Leaving directory '/home/evan/Documents/build/ghdl'
make: *** [Makefile:409: ghdl1-llvm] Error 2

tgingold, 2024-02-24:
Not too bad, you need to link with `-ldl`, just add `LDFLAGS=-ldl` to make or configure

evanfoss, 2024-02-24:
Thanks that works. It built but I think there is an unlisted dependency for make test. pyTooling? Seems like something I need to build as there isn't a package for it in debian.

__________________________________________________________________________ ERROR collecting lsp/LanguageServer.py __________________________________________________________________________
ImportError while importing test module '/home/evan/Documents/build/ghdl/testsuite/pyunit/lsp/LanguageServer.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.9/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
pyunit/lsp/LanguageServer.py:12: in <module>
    from pyGHDL.lsp.lsp import LanguageProtocolServer, LSPConn
../pyGHDL/__init__.py:59: in <module>
    from pyTooling.Decorators import export
E   ModuleNotFoundError: No module named 'pyTooling'

Paebbels, 2024-02-24:
The Python code (pyGHDL, LSP, DOM, ...) has various dependencies. These are all documented in:

https://github.com/ghdl/ghdl/blob/master/pyGHDL/requirements.txt
```
-r libghdl/requirements.txt
-r dom/requirements.txt
-r lsp/requirements.txt
-r cli/requirements.txt
```

https://github.com/ghdl/ghdl/blob/master/pyGHDL/lsp/requirements.txt
```
-r ../libghdl/requirements.txt
```

https://github.com/ghdl/ghdl/blob/master/pyGHDL/libghdl/requirements.txt
```
pyTooling >= 4.0.1, <5.0
```

So just run `pip install -r .../requirements.txt` and PIP installs these dependencies for you.  
(Assuming rasbian provides PIP.)

closed 2024-02-27 by tgingold

evanfoss, 2024-02-27:
Thanks that worked. 

So far the build process is
./configure --with-llvm-config  --prefix=/home/evan/opt/ghdl
make LDFLAGS=-ldl
make test
make install

I setup exports in my .bashrr

export PATH=/home/evan/opt/ghdl/bin:${PATH}
export LD_LIBRARY_PATH=/home/evan/opt/ghdl/lib:$LD_LIBRARY_PATH
export PKG_CONFIG_PATH=/home/evan/opt/ghdl/lib/pkgconfig:$PKG_CONFIG_PATH
export GHDL=/home/evan/opt/ghdl

Sadly the make test fails. It's not finding the libraries. I tried playing with the --enable-gplcompat and other options but it still fails. Is there another path I am missing?

Report issues on https://github.com/ghdl/ghdl
[GHDL - test] sanity
..Running with 4 test workers.
sanity 000hello: failed
analyze hello.vhdl
/home/evan/Documents/build/ghdl/ghdl_llvm:warning: ieee library directory '/home/evan/opt/ghdl/ieee/v93/' not found
error: cannot find "std" library
/home/evan/Documents/build/ghdl/ghdl_llvm: compilation error
sanity 002hello2008: failed
analyze hello.vhdl
/home/evan/Documents/build/ghdl/ghdl_llvm:warning: ieee library directory '/home/evan/opt/ghdl/ieee/v08/' not found
error: cannot find "std" library
/home/evan/Documents/build/ghdl/ghdl_llvm: compilation error
sanity 005examples: failed
analyze hello.vhdl
/home/evan/Documents/build/ghdl/ghdl_llvm:warning: ieee library directory '/home/evan/opt/ghdl/ieee/v93/' not found
error: cannot find "std" library
/home/evan/Documents/build/ghdl/ghdl_llvm: compilation error
sanity 000hello: failed
analyze hello.vhdl
/home/evan/Documents/build/ghdl/ghdl_llvm:warning: ieee library directory '/home/evan/opt/ghdl/ieee/v93/' not found
error: cannot find "std" library
/home/evan/Documents/build/ghdl/ghdl_llvm: compilation error

sanity 002hello2008: failed
analyze hello.vhdl
/home/evan/Documents/build/ghdl/ghdl_llvm:warning: ieee library directory '/home/evan/opt/ghdl/ieee/v08/' not found
error: cannot find "std" library
/home/evan/Documents/build/ghdl/ghdl_llvm: compilation error

sanity 005examples: failed
analyze hello.vhdl
/home/evan/Documents/build/ghdl/ghdl_llvm:warning: ieee library directory '/home/evan/opt/ghdl/ieee/v93/' not found
error: cannot find "std" library
/home/evan/Documents/build/ghdl/ghdl_llvm: compilation error

sanity test failed (000hello 002hello2008 005examples )
make: *** [Makefile:430: test.llvm] Error 1

evanfoss, 2024-02-27:
You were right. PIP worked and everything is great now. 

Thanks again for all the help.