ghdl/ghdl #2614: testbench signal transitions not visible in FST if signal is input into recursive RTL

jeras, 2024-02-24:
Git version of GHDL.

I am not sure the RTL recursion is to blame here, but I did test with two different recursive implementations, and there were no issues without the recursive RTL.

Signal `xi` should be toggling simultaneously with `cnt`, instead it seems to be stuck at zero.
Interestingly, the simulation seems to work just fine, the RTL output is toggling, jut the waveform dump is wrong.

[image]

I attached the VHDL source files, for a regression test.

[recursive_xor.zip](https://github.com/ghdl/ghdl/files/14394328/recursive_xor.zip)

closed 2024-02-25 by tgingold

jeras, 2024-02-25:
Thanks, it works on the rest of my code too.

tgingold, 2024-02-25:
Great!  Thank you for the status update.

commit 41752af6 (2024-02-25): simul-fst: fix handling of sub-aliases

Fix #2614