ghdl/ghdl #2618: Integer literal overflow

leonschoorl, 2024-02-29:
**Description**
From what I understand VHDL requires `integer` to support at least the range -(2^31 - 1)  to 2^31 - 1,  or -2147483647 to 2147483647.
And running `ghdl --disp-standard` shows ghdl should give me one more negative integer on top of that:
```vhdl
package standard is
  [...]
  type integer is range -2147483648 to 2147483647;
  [...]
end;
```

But it looks like ghdl doesn't allow me to write integer literal of -2147483600 or smaller.
And `-2147483600 == -(2^31 - 1) + 48`.
One bigger, -2147483599, is accepted.

ghdl reports:
```
ent.vhdl:6:46:error: literal overflow
  constant underflows : integer := -2147483600;
                                             ^
ent.vhdl:6:46:error: space is required between numbers
  constant underflows : integer := -2147483600;
                                             ^
ent.vhdl:6:46:error: missing ";" at end of object declaration
  constant underflows : integer := -2147483600;
                                             ^
ghdl:error: compilation error
```

**Expected behaviour**
The integers within range to be accepted without errors or warnings.

**How to reproduce?**
```vhd :file: ent.vhd
entity ent is
end;

architecture structural of ent is
  constant stillOk    : integer := -2147483599;
  constant underflows : integer := -2147483600;
begin
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
```

**Context**
OS: Linux

The lastest version I've tested this with is `GHDL 4.0.0-dev (v3.0.0-558-g8a9eb7e25)` from the docker image `ghdl/ghdl:ubuntu22-llvm-11`.
But `GHDL 3.0.0` and ``GHDL 2.0.0` show the same behavior.

leonschoorl, 2024-02-29:
The upper bound of the range is where I would expect.
But interestingly the message reported for going over is different:
```
entPos.vhdl:6:35:warning: static expression violates bounds [-Wruntime-error]
  constant overflows : integer := 2147483648;
                                  ^
```

And it's a warning, not an error.

closed 2024-02-29 by tgingold

leonschoorl, 2024-03-01:
Ah, so the offset of 48 was the numeric value of the character `0`.

Thanks for the quick fix

tgingold, 2024-03-02:
You're welcome.  I thought the parentheses were correctly set everywhere, but...

commit 053b4baa (2024-02-29): vhdl-scanner: avoid an overflow

Fix #2618