ghdl/ghdl #2619: GHDL unable to load package std_logic_1164

KWJ314, 2024-03-03:
Hello,

I'm certain this is an issue on my part here, likely me misunderstanding something, but I ran into a snag when using GHDL to learn VHDL and I am unsure where else to go.  I am working through some tutorials online and I have just come to the point of using the std_logic type, but when I try to import that particular library, I get this error:

C:\>ghdl -a --ieee=standard T10_StdLogicTb.vhdl
T10_StdLogicTb.vhdl:3:10: cannot load package "std_logic_1164"
use IEEE.std_logic_1164.all;
         ^
C:\msys64\mingw64\bin\ghdl.exe: compilation error

I cannot find anything in the documentation that implies that I need to do anything special in GHDL to use it, so I am a bit confused as to what is causing this. I can see the std_logic_1164 in the background files, so I do not think it is an install issue. If it helps, this was installed on a Windows 10 64 bit system using msys2.

Below is the code that I'm trying to analyze. Again, I realize I'm probably doing something simple wrong, but any help would be greatly appreciated:

library IEEE;
use IEEE.std_logic_1164.all;

entity T10_StdLogicTb is
end entity;

architecture sim of T10_StdLogicTb is

	signal Signal1 : std_logic := '0';
	signal Signal2 : std_logic;
	signal Signal3 : std_logic;

    -- TODO: Declare a signal named "Signal1"
    --       of type std_logic with default value '0'
    
    -- TODO: Declare a signal named "Signal2" of type std_logic
    
    -- TODO: Declare a signal named "Signal3" of type std_logic

begin

    process is
    begin
        
        wait for 10 ns;
        -- TODO: Invert the value of Signal1
		Signal1 <= not Signal1;

    end process;

    -- Driver A
    process is
    begin

        -- TODO: Constantly drive 'Z' to Signal2
        
        -- TODO: Constantly drive '0' to Signal3
		
		Signal2 <= 'Z';
		Signal3 <= '0';
        wait;

    end process;

    -- Driver B
    process(Signal1) is
    begin

        -- TODO: Drive 'Z' to both Signal2 and Signal3 if Signal1 is '0'
        --       Drive '1' to both Signal2 and Signal3 if Signal1 is not '0'
        --       
        --       Inspect the signals in the waveform and verify that the
        --       values are according to the resolution table.
        --       The resolution table is found here:
        --       https://vhdlwhiz.com/std_logic/
		if Signal1 = '0' then
			Signal2 <= 'Z';
			Signal3 <= 'Z';
		else
			Signal2 <= '1';
			Signal3 <= '1';
		end if;
    
    end process;

end architecture;

tgingold, 2024-03-04:
It's probably an installation issue.  So, first, how did you install ghdl ?  What is the version ('ghdl --version') ?

Then, what is the output of 'ghdl --disp-config' ?

You can also remove '--ieee=standard', as this is the default.

KWJ314, 2024-03-04:
Hello Tgingold,

Thank you for the help I installed ghdl through msys2, specifically through the mingw-w64-x86_64-ghdl-llvm binary package (if you were to ask me why that one and not the mingw-w64-ucrt-x86_64-ghdl-llvm package, I'm afraid I can't give you a better answer than it was the first one). I used that since it seemed to be the easiest.

The version given from GHDL is:
GHDL 4.0.0-dev (3.0.0.r750.g2135cbf14) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 llvm 17.0.6 code generator.

The config output is:
Paths at configuration:
compiler command: ghdl1-llvm.exe
linker command (--LINKER=, CC, or cc): cc
default lib prefix: lib

command line prefix (--PREFIX): (not set)
environment prefix (GHDL_PREFIX): (not set)
exec prefix (from program name): C:\msys64\mingw64

library prefix: C:\msys64\mingw64\lib/ghdl
library directory: C:\msys64\mingw64\lib/ghdl
compiler path: C:\msys64\mingw64\bin\ghdl1-llvm.exe
linker path: C:\msys64\mingw64\bin\cc.exe

default library paths:
 C:\msys64\mingw64\lib/ghdl\ieee\v93\
 C:\msys64\mingw64\lib/ghdl\

Looking at that, I wonder if the different slashes might have something to do with it (it would be weird, but I have seen weirder things).

Thanks again for the help. And thanks for the tip on the ieee flag. I added it to make sure that it wasn't the cause of the error when I first saw it, but it will be nice to not have to type it in every time going forward.

tgingold, 2024-03-05:
Did you check the files are present in `C:\msys64\mingw64\lib/ghdl\ieee\v93` and *readable* ?

KWJ314, 2024-03-06:
Hey Tgingold,

I did double check the file path and there are certainly files in there. I guess I don't know what you mean exactly by readable. They are all object files, so most of it isn't human readable (though there are a few lines of text in there). If you mean readable by GHDL, I guess I don't know how to double check, but I did make sure that there aren't any permissions on that folder.

tgingold, 2024-03-06:
Ah ah, so the source files are missing.  Maybe a missing dependency should be installed.  

You can also try to install prepackages binaries from https://github.com/ghdl/ghdl/releases/tag/nightly

KWJ314, 2024-03-07:
Looking over it again, I either missed or neglected to mention (I have no idea why) 1 file that was in that default library folder. Specifically there is a ieee-obj93.cf file in there. That seems to be a text file with format like:

"file . "../../src/ieee/v93/std_logic_1164.vhdl" "89fb289e40b09502e16490e5077b20678734f167" "20231221115424.940":
  package std_logic_1164 at 54( 2597) + 0 on 11 body;"

There are source vhdl files in the directory indicated in the first line, incidentally, including std_logic_1164.vhdl.

I suppose one thing that I do not understand, and may help find what dependency I am missing or other file I need from that repository, is what is GHDL looking for in that library folder? Is it looking for those object files? Is it the VHDL files? Am I missing something completely (which is probably the case to be honest).

Thanks again.

diogratia, 2024-03-07:
The corresponding entry of the /usr/local/lib/ghdl/ieee/v93/ieee-93obj.cf file for a recent nightly build:

```none
file . "../../src/ieee/v93/std_logic_1164.vhdl" "89fb289e40b09502e16490e5077b20678734f167" "20240307062606.422":
  package body std_logic_1164 at 54( 2590) + 0 on 12;
```
The access to source file is through a hierarchical path relative to the ieee-93obj.cf configuration file which specifies which source file was analyzed to create the matching object file. Tristan has established rules for how the object file will be named in the library which is collectively the object files and here the ieee-93obj.cf configuration file.

The error message comes from ghdl/src/vhdl/vhdl-sem_lib.adb line 208 if your were to examine the ghdl src code. That's in procedure Load_Parse_Design_Unit which begins at line 184. The locations identified by find traversing Ada source and executing grep "cannot load" in a ghdl source tree.

The VHDL source file is parsed again as a means of making it's declarations directly visible.  It's less implementation complexity than storing symbol tables with a large number of attributes and reassigning object IDs when loading symbol tables. The source can be required for error messages in any event.

The GMT date time stamp in the configuration file and modification date of the source file are subsequently compared. There's a requirement that any object in a reference library be reanalyzed if it's source file has changed. This has the added benefit of allowing the source file to be parsed again.

The error message is telling you that source file cannot be read. That would be because the hierarchical relationship between the location of the configuration file and the source file has changed or the source file isn't present or the source file can't be read for the usual host operating system (execution environment) reasons.

One thing you could do when installing ghdl is to remove any preexisting ghdl installation unless the installation locations are mutually exclusive. There is no installation mechanism to remove existing files and you can run into permission issues. 

A ghdl installation is relocatable, either finding it's needed files from the executable location (ghdl1-llvm.exe in your case) or by relative hierarchical relationship in the case of the library configuration files. You can simply move a preexisting installation aside. You can extract a github ghdl binary distribution to a particular target location as well. The pathnames in the .zst archives (in your case) are not absolute.

To understand where the loading of source for package std_logic_1164 goes awry you could explore ghdl's source code, examining the code preceding line 208 in vhdl-sem_lib.adb either through a debugger or by by checking out the source code and examining the various subprograms preceding the line 208 assertion. 

In the mean time ghdl-4.0 was released yesterday noting an llvm distribution can depend on dynamic libraries provided by an llvm compiler installation. In general it's unsafe to try to use libraries (libLLVM) from another llvm major version. The latest nightly build uses llvm-17.0.6.

KWJ314, 2024-03-11:
Hello Diogratia and Tgingold,

Thank you both for the help. I finally got it working. I did initially have some luck by replacing the cf file, but that eventually lead to other issues. Eventually I uninstalled and reinstalled, and that seemed to resolve it. I thought I had tried that earlier, but this was over a week ago and at this point I can't remember.

Anyway, thank you again for the help. Best regards.

closed 2024-03-11 by KWJ314