ghdl/ghdl #2621: --no-run ignored on ghdl 4.0.0-dev

carabsc, 2024-03-08:
I've been using the following script (abbreviated) to generate a signal list from some VHDL designs:

```
ghdl -i --workdir=<work directory> <list of files in design>

ghdl -m -P<vendor library> --no-vital-checks --syn-binding -fexplicit --ieee=synopsys --workdir=<work directory> <top module>

ghdl -r -P<vendor library> --no-vital-checks --syn-binding -fexplicit --ieee=synopsys --workdir=<work directory> <top module> --no-run --write-wave-opt=<signals file> --vcd=<vcd file>
```

This worked perfectly with ghdl 1.0.0 but with 4.0.0-dev the simulation run never stop, ignoring the --no-run option, unless I provide a --stop-time option. The software was built from Git code, using the mcode backend.

closed 2024-03-10 by tgingold

carabsc, 2024-03-11:
A clarification is needed regarding what is supposed to happen with this command line:

`ghdl -r -P<vendor library> --no-vital-checks --syn-binding -fexplicit --ieee=synopsys --workdir=<work directory> testbench --no-run --stop-time=1ns --write-wave-opt=<signal file> --vcd=<VCD file>`

As I mentioned initially GHDL v1.0.0 would generate a signal file containing all the signals in the design. Running GHDL v4.0.0-dev with this proposed patch creates the signal file but it is empty.
In my view this is a regression, but maybe this was never supposed to work on 1.0.0.

tgingold, 2024-03-12:
Yes, it makes sense to allow`--write-wave-opt`.

commit 2cd16ff2 (2024-03-10): simul-main: handle --no-run

Fix #2621