ghdl/ghdl #2622: Why is the :error: severity marker only output to stderr if outputing to a terminal in Linux ?

carabsc, 2024-03-09:
GHDL uses ANSI codes to color the stderr output, highlighting the severity of the logged message, if supported by the running environment. All severity markers are always output as part of the logged message, except for :error: which is only output if the environment supports coloring (which currenly means running on a Linux terminal). Since GHDL also outputs context info together with warnings it makes it impossible to discern the severity of the message from captured stderr, as context info lines have to be considered errors. Is there a reason for this behavior ?

From src/errorout-console.adb:

```
         when Msgid_Error =>
            if Flag_Color_Diagnostics = On then
               Set_Color (Color_Error);
            end if;
            if Msg_Len = 0
              or else Flag_Color_Diagnostics = On
            then
               --  'error:' is displayed only if not location is present, or
               --  if messages are colored.
               Put ("error:");
            end if;
         when Msgid_Fatal =>
            if Flag_Color_Diagnostics = On then
               Set_Color (Color_Fatal);
            end if;
            Put ("fatal:");
```

tgingold, 2024-03-10:
This behaviour is somewhat historical and followed the gnu standard.  Maybe you'd like to have `error:` when there are context lines ?

carabsc, 2024-03-11:
No, I just don't understand why should the output of the marker be conditioned on it being done to a tty. In my view it should always be output. Currently if the program is run on a Linux console the error lines are formatted as:

`<file>:<line>:<column>:<error marker colored in red>: <error message>
`

while the same error lines when captured through stderr are formatted as:

`<file>:<line>:<column>: <error message>
`

That's what I don't get and complicates parsing the output from the program.

tgingold, 2024-03-12:
Ok, I will try to always emit them (or at least when there are context).

commit cebce322 (2024-03-19): errorout-console: disp error marker if caret diagnostics

Fix #2622
closed 2024-03-19 by tgingold