ghdl/ghdl #2625: VHDL 2008 hex notation with defined number of bits

arretxe, 2024-03-15:
**Description**
VHDL 2008 allows writting in hex format a number of bits not proportional to 4, by appending the number of bits prior to the x char, like in this example:
 constant LBR_BCK_STS_AR   : std_logic_vector(16 downto 0) := 17x"1-000";  -- "1----000000000000"

However GHDL does not seem to recognize this:
lb_if_pkg.vhd:121:63: invalid character in bit string  constant LBR_BCK_STS_AR  : std_logic_vector(16 downto 0)  := 17x"1-000";

**Context**

- OS: Windows UCRT64
- Origin:
  - [ ] Released binaries: tried with v4.0.0 and nightly

umarcor, 2024-03-15:
@arretxe, please provide a complete reproducer:

- A complete file with an entity and an architecture which we can use for testing.
- The specific commands you used.

I just tested the following and it works:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ent is
end;

architecture arch of ent is
constant LBR_BCK_STS_AR : std_logic_vector(16 downto 0) := 17x"1-000"; -- "1----000000000000"
begin
  process begin
    report to_hstring(LBR_BCK_STS_AR);
    wait;
  end process;
end architecture;
```

```
# ghdl -a --std=08 ent.vhd 
# ghdl -e --std=08 ent
# ./ent
ent.vhd:11:5:@0ms:(report note): 1X000
# ghdl version
GHDL 4.0.0-dev (3.0.0.r750.g2135cbf14) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 llvm 17.0.6 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I believe you might have forgotten to specify `--std=08`, so the code is being analysed as VHDL <2008.

arretxe, 2024-03-15:
I actually do specify --std=08 but I forgot to mention I use ghdl through YOSYS, does that influence?

My command actually looks like this:
C:\msys64\mingw64\bin\yosys.exe -p "ghdl -fsynopsys 
 --work=unisim
      C:\Xilinx\Vivado\2022.1\data\vhdl\src\unisims\unisim_VCOMP.vhd
      C:\Xilinx\Vivado\2022.1\data\vhdl\src\unisims\unisim_VPKG.vhd
    --work=work
 --std=08
 ..\..\..\../FCPUD/LB_IF/Src/fcpud_lb_if_pkg.vhd
 ..\..\..\../many_more_vhd_files.vhd
 -e; ; proc; ; write_json teroshdl_yosys_output2.json; stat"

tgingold, 2024-03-16:
can you try to create a reproducer (maybe starting from umarcor example) ?

closed 2024-04-29 by tgingold