ghdl/ghdl #2626: raised CONSTRAINT_ERROR : trans-chap4.adb:3427 access check failed

MJoergen, 2024-03-15:
**Description**
GHDL crashes when running a simulation.

**Expected behaviour**
It should not crash. When running the exact same simulation using QuestaSim, there are no compilation errors, and the simulation runs as expected.

**How to reproduce?**
See attached file
[ghdl_crash.zip](https://github.com/ghdl/ghdl/files/14614888/ghdl_crash.zip)

In the directory `ghdl_crash/sim` run the following command
```
ghdl -c --std=08 ../src/Example_Design/duplicator.vhd ../src/Example_Design/hexifier.vhd ../src/Example_Design/serializer.vhd ../src/Example_Design/merginator.vhd ../src/SDCard/sdcard_globals.vhd ../src/SDCard/sdcard_cmd.vhd ../src/Example_Design/sdcard_cmd_logger.vhd ../src/SDCard/sdcard_dat.vhd ../src/SDCard/sdcard_ctrl.vhd ../src/SDCard/sdcard_wrapper.vhd sdcard_sim.vhd tb_sdcard.vhd -r tb_sdcard  --assert-level=error --wave=tb_sdcard.ghw --stop-time=200us
```

**Context**
Please, provide the following information:

- OS: "Linux Mint 21.1 Vera"
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `ec7cd5909`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.0.0.r5.gec7cd5909) [Dunoon edition]
Compiled with GNAT Version: 10.5.0
Target: x86_64-linux-gnu
/home/mike/ghdl_crash/sim/
Command line:
ghdl -c --std=08 ../src/Example_Design/duplicator.vhd ../src/Example_Design/hexifier.vhd ../src/Example_Design/serializer.vhd ../src/Example_Design/merginator.vhd ../src/SDCard/sdcard_globals.vhd ../src/SDCard/sdcard_cmd.vhd ../src/Example_Design/sdcard_cmd_logger.vhd ../src/SDCard/sdcard_dat.vhd ../src/SDCard/sdcard_ctrl.vhd ../src/SDCard/sdcard_wrapper.vhd sdcard_sim.vhd tb_sdcard.vhd -r tb_sdcard --assert-level=error --wave=tb_sdcard.ghw --stop-time=200us
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap4.adb:3427 access check failed
******************************************************************
```

diogratia, 2024-03-15:
This is with the llvm back end version of the same release:

```none
%: ghdl -c --std=08 ../src/Example_Design/duplicator.vhd ../src/Example_Design/hexifier.vhd ../src/Example_Design/serializer.vhd ../src/Example_Design/merginator.vhd ../src/SDCard/sdcard_globals.vhd ../src/SDCard/sdcard_cmd.vhd ../src/Example_Design/sdcard_cmd_logger.vhd ../src/SDCard/sdcard_dat.vhd ../src/SDCard/sdcard_ctrl.vhd ../src/SDCard/sdcard_wrapper.vhd sdcard_sim.vhd tb_sdcard.vhd -r tb_sdcard  --assert-level=error --wave=tb_sdcard.ghw --stop-time=200us
ghdl:error: cannot open -r
ghdl:error: cannot open tb_sdcard
ghdl:error: cannot open --assert-level=error
ghdl:error: cannot open --wave=tb_sdcard.ghw
ghdl:error: cannot open --stop-time=200us
%: ghdl -a --std=08 ../src/Example_Design/duplicator.vhd ../src/Example_Design/hexifier.vhd ../src/Example_Design/serializer.vhd ../src/Example_Design/merginator.vhd ../src/SDCard/sdcard_globals.vhd ../src/SDCard/sdcard_cmd.vhd ../src/Example_Design/sdcard_cmd_logger.vhd ../src/SDCard/sdcard_dat.vhd ../src/SDCard/sdcard_ctrl.vhd ../src/SDCard/sdcard_wrapper.vhd sdcard_sim.vhd tb_sdcard.vhd
%: ghdl -e --std=08 tb_sdcard
%: ghdl -r tb_sdcard --assert-level=error --wave=tb_sdcard.ghw --stop-time=200us
tb_sdcard.vhd:56:7:@140ns:(report note): Test finished
%: ghdl --version
GHDL 5.0.0-dev (4.0.0.r5.gec7cd5909) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: which make
/usr/bin/make
%: make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
%: 
```

Which appears to say that the make sim makefile issued command is incompatible with ghdl, apparently differently between the two platforms or  different back end versions. The same occurs when following a make clean with a make sim. A make -n sim produces the command line in your issue.

Running analysis, elaboration and simulation commands separately with partitioned command line arguments succeeds.

When successfully run with separate analysis, elaborate and simulate command invocations we see that the  simulation doesn't appear to do anything interesting as yet:

[image]

which likely means you see this as a roadblock which can likely be alleviated by changing your makefile to use  separate commands as a work around.

Whether there's an issue with your command line, and a lack of safe error reporting or executing the command line isn't readily apparent without pouring through either code or ghdl documentation. I'm on the wrong side of the International Dateline, haven't had a caffeine fix, have to go feed horses and the days are getting shorter here in the southern hemisphere.

[image]

tgingold, 2024-03-18:
I can also reproduce the crash.

MJoergen, 2024-03-18:
@diogratia Thank you for your detailed analysis! That's very helpful and enlightening.

So, while waiting for a fix, there are two possible workarounds (either of which can be used):
* Use the llvm backend (instead of the mcode version).
* Avoid using non-static expressions in port maps.

Is that correctly understood?

MJoergen, 2024-03-18:
I can confirm that using the llvm backend works as described by @diogratia , i.e. no crash.

commit 666012ed (2024-03-21): translate: skip unused generate statement body

for pre-elaboration.

Fix #2626
closed 2024-03-21 by tgingold