ghdl/ghdl #2629: ghdl 4.0.0 fails to build on powerpc64le-linux-gnu

doko42, 2024-03-21:
seen, when trying to build ghdl 4.0.0 using GCC 13

/<<PKGBUILDDIR>>/builddir/gcc/build/gcc/ghdl -a --GHDL1=/<<PKGBUILDDIR>>/builddir/gcc/build/gcc/ghdl1 --std=87 --bootstrap --work=std ../../src/std/v87/textio.vhdl
during RTL pass: final
/<<PKGBUILDDIR>>/builddir/gcc/lib/ghdl/gcc/vhdl/std/v87/../../src/std/v87/textio.vhdl: In function ‘std__textio__ELAB_SPEC’:
/<<PKGBUILDDIR>>/builddir/gcc/lib/ghdl/gcc/vhdl/std/v87/../../src/std/v87/textio.vhdl:1: internal compiler error: in rs6000_output_function_epilogue, at config/rs6000/rs6000-logue.cc:5371
    1 | --  Std.Textio package declaration.  This file is part of GHDL.
      | 
0x111da203 rs6000_output_function_epilogue(_IO_FILE*)
	../../src/gcc/config/rs6000/rs6000-logue.cc:5371
0x108920bb final_end_function()
	../../src/gcc/final.cc:1860
0x10897a53 rest_of_handle_final
	../../src/gcc/final.cc:4248
0x10897a53 execute
	../../src/gcc/final.cc:4318
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.

complete build log at
https://launchpad.net/ubuntu/+source/ghdl/4.0.0+dfsg-0ubuntu1/+build/27947473

doko42, 2024-03-21:
GCC report: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=114420

tgingold, 2024-03-22:
Did you check https://github.com/ghdl/ghdl/blob/master/doc/development/building/GCC.rst ?  (the last paragraph).

doko42, 2024-03-24:
thanks, package built for Ubuntu 24.04 LTS:
https://launchpad.net/ubuntu/+source/ghdl/4.0.0+dfsg-0ubuntu1

closed 2024-03-24 by doko42