ghdl/ghdl #2630: [BUG] Cannot use external names with constants

Jonas4242, 2024-03-25:
**Description**
It seems like you cannot reference constants with the external names mechanism. I want to use this in my test bench so I do not have to create a package for the constants.

**Expected behaviour**
I expect to have access to the external constant and its value, but instead, the simulation fails with an error: `NULL access dereferenced`.

**How to reproduce?**
Here is a MWE:

```vhd :file: ext_name.vhd
use std.textio.all;

entity ext_name is
end ext_name;

architecture behaviour of ext_name is
begin
  a : block is
    constant A_C : natural := 42;
  begin
  end block;

  b : block is
    alias A_C is << constant .ext_name.a.A_C : natural >>;
  begin
    b_proc : process
      variable l : line;
    begin
      write (l, to_string(A_C));
      writeline (output, l);
      wait;
    end process;
  end block;
end behaviour;
```

```sh :image: ghdl/ghdl:bullseye-gcc
ghdl -a --std=08 ext_name.vhd
ghdl -e --std=08 ext_name
./ext_name
```
[image]

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04 LTS
-  I used a gcc-backend GHDL from the nightly binaries for Ubuntu 22.04:
  - GHDL 5.0.0-dev (4.0.0.r14.gf845482a2) [Dunoon edition]  Compiled with GNAT Version: 10.5.0
  - GCC 12.1.0 code generator

tgingold, 2024-03-25:
You should test with ghdl with mcode backend.  External names are not supported with the gcc backend.

abyszuk, 2024-03-26:
Hi @tgingold, thanks for the answer. Is there a plan to make them supported or we need to come up with a workaround?

diogratia, 2024-03-27:
It fails with the mcode version as well:

```none
%: $GHDL -r --std=08 ext_name

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.0.0.r5.gec7cd5909) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/dio_gratia/Desktop/
Command line:
/Users/dio_gratia/opt/share/4.0.0.r5.gec7cd5909/ghdl-macos-11-mcode/bin/ghdl -r --std=08 ext_name
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : elab-vhdl_context.adb:707 access check failed
******************************************************************
%: 
```

tgingold, 2024-03-27:
No, there is no plan to support external names with the gcc backend ?
Do you really need the gcc backend ?  If it is for coverage, it is now available with mcode backend (option `--coverage` and command `coverage` to merge results). I will propose a PR for vunit soon.

tgingold, 2024-03-27:
@diogratia Yes indeed,  I am investigating.

abyszuk, 2024-03-27:
> No, there is no plan to support external names with the gcc backend ? Do you really need the gcc backend ? If it is for coverage, it is now available with mcode backend (option `--coverage` and command `coverage` to merge results). I will propose a PR for vunit soon.

Yes, we need the GCC backend for coverage.
It's fantastic news that this is coming to `mcode` backend too! Is support for LLVM backend planned as well?

closed 2024-03-27 by tgingold

tgingold, 2024-03-28:
LLVM will become a jit backend like mcode, so yes.

commit 017ad406 (2024-03-27): simul-vhdl_elab: fix handling of constant external names

Fix #2630