ghdl/ghdl #2633: [Bug] - pyGHDL: error when importing pyGHDL packages

pidgeon777, 2024-03-27:
Referenced also here:

https://github.com/VHDL/pyVHDLModel/issues/77#issue-2211103727

I tried to run this code:

```python
from pathlib import Path
from pyGHDL.dom.NonStandard import Design, Document
```

but the second import crashes:

```log
(common) C:\Work\Scripts\Python\PyGHDL>python test.py        
Traceback (most recent call last):
  File "C:\Work\MEGA\Scripts\Python\PyGHDL\test.py", line 2, in <module>
    from pyGHDL.dom.NonStandard import Design, Document
  File "C:\Users\<USERNAME>\scoop\apps\miniconda3\current\envs\common\Lib\site-packages\pyGHDL\dom\NonStandard.py", line 43, in <module>
    from pyGHDL.dom.Names import SimpleName
  File "C:\Users\<USERNAME>\scoop\apps\miniconda3\current\envs\common\Lib\site-packages\pyGHDL\dom\Names.py", line 37, in <module>
    from pyVHDLModel.Name import Name
  File "C:\Users\<USERNAME>\scoop\apps\miniconda3\current\envs\common\Lib\site-packages\pyVHDLModel\__init__.py", line 60, in <module>
    from pyTooling.Graph           import Graph, Vertex, Edge
  File "C:\Users\<USERNAME>\scoop\apps\miniconda3\current\envs\common\Lib\site-packages\pyTooling\Graph\__init__.py", line 63, in <module>
    from pyTooling.Tree        import Node
  File "C:\Users\<USERNAME>\scoop\apps\miniconda3\current\envs\common\Lib\site-packages\pyTooling\Tree\__init__.py", line 100, in <module>
    class Node(Generic[IDType, ValueType, DictKeyType, DictValueType], metaclass=ExtendedType, useSlots=True):
  File "C:\Users\<USERNAME>\scoop\apps\miniconda3\current\envs\common\Lib\site-packages\pyTooling\MetaClasses\__init__.py", line 250, in __new__
    members['__slots__'] = self.__getSlots(baseClasses, members)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\<USERNAME>\scoop\apps\miniconda3\current\envs\common\Lib\site-packages\pyTooling\MetaClasses\__init__.py", line 431, in __getSlots
    raise AttributeError(f"Base-class '{base.__name__}' has no '__slots__'.")
AttributeError: Base-class 'Generic' has no '__slots__'.
```

I will provide any additional information if needed.

pidgeon777, 2024-03-27:
`pip list` output:

```log
Package                 Version
----------------------- --------
archspec                0.2.1
argcomplete             3.2.3
boltons                 23.0.0
certifi                 2024.2.2
cffi                    1.16.0
charset-normalizer      2.0.4
colorama                0.4.6
conda                   24.1.0
conda-content-trust     0.1.3
conda-libmamba-solver   24.1.0
conda-package-handling  2.2.0
conda_package_streaming 0.9.0
cryptography            39.0.1
distro                  1.8.0
gitdb                   4.0.11
GitPython               3.1.42
idna                    3.4
jsonpatch               1.32
jsonpointer             2.1
libmambapy              1.5.6
menuinst                2.0.2
packaging               23.1
pip                     23.1.2
platformdirs            3.10.0
pluggy                  1.0.0
pyAttributes            2.5.1
pycosat                 0.6.6
pycparser               2.21
pyGHDL                  4.0.0
pyTooling               4.0.1
pyVHDLModel             0.25.1
requests                2.31.0
ruamel.yaml             0.17.21
ruamel.yaml.clib        0.2.6
setuptools              67.8.0
six                     1.16.0
smmap                   5.0.1
tqdm                    4.65.0
truststore              0.8.0
urllib3                 2.1.0
wheel                   0.38.4
zstandard               0.19.0
```

chenailuoding1, 2024-11-24:
Have you solved this problem, I have the same problem, can you communicate with you

Paebbels, 2024-11-24:
A few days ago, tgingold merge a new pipeline and also fixes for pyGHDL from my branch. This should solve some of the mentioned problems.

closed 2024-11-24 by Paebbels