ghdl/ghdl #2638: [Bug] GHDL fails to analyze the `mti_util.vhd` ModelSim library file

pidgeon777, 2024-04-04:
**Description**

**GHDL** fails to analyze the **ModelSim** `mti_util.vhd` library file.

**Expected behaviour**

I would expect **GHDL** to analyze the **ModelSim** library source without any issues.

**How to reproduce?**

`mti_util.vhd` content:

```vhd :file: mti_util.vhd
--
-- Copyright 1991-2023 Mentor Graphics Corporation
--
-- All Rights Reserved.
--
-- THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION WHICH IS THE PROPERTY OF 
-- MENTOR GRAPHICS CORPORATION OR ITS LICENSORS AND IS SUBJECT TO LICENSE TERMS.
--  
-- --------------------------------------------------------------------

package util is

-----------------------------------------------------------------------
-- SIGNAL SPY 
-----------------------------------------------------------------------
  type del_mode is (MTI_INERTIAL, MTI_TRANSPORT);
  type forcetype is (default, deposit, drive, freeze);
  alias mti_default is default [return forcetype];         -- Used for VHDL2008 compatability
  attribute builtin_subprogram : string;
        
  procedure disable_signal_spy (
                     source_signal      : IN string ;
                     destination_signal : IN string ;
                     verbose            : IN integer := 0);
       attribute builtin_subprogram of disable_signal_spy : procedure is "disable_signal_spy_vhdl";

  procedure enable_signal_spy (
                     source_signal      : IN string ;
                     destination_signal : IN string ;
                     verbose            : IN integer := 0);
       attribute builtin_subprogram of enable_signal_spy : procedure is "enable_signal_spy_vhdl";

  procedure init_signal_spy (
                     source_signal      : IN string ;
                     destination_signal : IN string ;
                     verbose            : IN integer := 0;
                     control_state      : IN integer := -1);
       attribute builtin_subprogram of init_signal_spy : procedure is "init_signal_spy_vhdl_wc";

  procedure init_signal_driver (
                     source_signal      : IN string ;
                     destination_signal : IN string ;
                     delay              : IN time := 0 ps;
                    delay_mode         : IN del_mode := MTI_INERTIAL;
                     verbose            : IN integer := 0) ;
       attribute builtin_subprogram of init_signal_driver : procedure is "init_signal_driver_vhdl";

  procedure signal_force (
                     destination_signal : IN string ;
                     force_value        : IN string;
                     force_delay        : IN time := 0 ps;
                     force_type         : IN forcetype := default;
                     cancel_period      : IN time := -1 ms;
                     verbose            : IN integer := 0) ;
       attribute builtin_subprogram of signal_force : procedure is "signal_force_vhdl";

  procedure signal_release (
[... 468 lines cut ...]

It seems that one of the errors is that the `default` keyword is used as a type value here:

```vhdl
  type forcetype is (default, deposit, drive, freeze);
```

I wonder why this is not supported by GHDL?

**Context**

- OS: Windows 11
- Origin:
  - Released binaries. GHDL Version:

```log
GHDL 4.0.0-dev (3.0.0.r871.g45fcff5bc) [Dunoon edition]
 Compiled with GNAT Version: 13.2.0
 llvm 17.0.6 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tmeissner, 2024-04-04:
>  It seems that one of the errors is that the default keyword is used as a type value here:
>
>  `type forcetype is (default, deposit, drive, freeze);`
>
> I wonder why this is not supported by GHDL?

Because `default` is a reserved keyword since VHDL-2008. It's used by the integrated PSL language für defining the default clock:

```vhdl
default clock is rising_edge(clk);
```

pidgeon777, 2024-04-04:
@tmeissner Thanks for the explanation. I wonder if there is a way in GHDL to make this error a warning instead?

This is because it seems that **ModelSim** still compiles the file even when using the reserved keyword, leading to an error instead.

Also in some other VHDL source files we use the reserved word "DEFAULT":

```vhdl
type tDESIGN is (DEFAULT, DESIGN1, DESIGN2, DESIGN3); -- design name
```

and Vivado is able to compile the file without any problems.

Or, more generally, how could this **ModelSim** file be compiled with GHDL?

pidgeon777, 2024-04-05:
I'm wondering if there is a way to suppress this error with a flag, or convert it to a warning.

I read there are these flags:

[--warn-reserved](https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-warn-reserved "Permalink to this definition")

[-Wreserved](https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-Wreserved "Permalink to this definition")

> Emit a warning if an identifier is a reserved word in a later VHDL standard.

I'm not sure if those would help, though. What do you think, @tgingold?

`nvc`, the other VHDL simulator, reports this warning: `'DEFAULT' is a reserved word in VHDL-2008`.

diogratia, 2024-04-06:
The "shall not" in 15.10 is imperative (mandatory, see 1.3 Structure and terminology of this standard, 1.3.1 General). 

The warning is to discourage users analyzing a VHDL design file under a previous revision of the standard from using a reserved word in a later revision of the standard and creating the situation you find yourself in. The idea being part of your tool chain being limited to compliance with an older revision with some possibility the use of a newer affected version of the standard might be possible in the future.

 Not all VHDL implementations are capable of mixing libraries or library objects from different revisions of the standard. You could note the prefix for a ghdl .cfg file contains an indication of the VHDL revision of the library it specifies.

Also note the first year in the copyright range in the file shown in your issue predating the -2008 revision.

On top of all that the various calls to FOREIGN subprograms that don't exist in ghdl would preclude functionality should you successfully analyze these files. You'd get those error messages you see in the required subprogram bodies you are required to supply and execute when the FOREIGN subprograms are not  made available during elaboration telling you the calls failed.

Without the functionality behind the APIs you'd only be demonstrating those three packages analyze without reaping any benefit for verification.

You could note that the trade secret claims in the file header makes it difficult to reverse engineer the MTI calls behind the package subprograms in some venues because of recent shifts in burden of proof and relaxing duty of care by the trade secret owner. If you don't have clean hands for being exposed to the API you'd be at risk without someone clean room documenting an equivalent API for the package subprograms targeting a distinct VHDL implementation.  

Some of the functionality in these packages is also starting to creep into the standard (VHPI, -2019). It may not be worth it to tool developers to pursue for purposes of legacy design portability.

pidgeon777, 2024-04-06:
Thanks for the very detailed and informative reply 👍.

Regarding `nvc', it seems that the warning occurs when compiling with the `1993` flag. When compiling with `2008`, the warning is reported as an error instead.

tgingold, 2024-04-09:
You can try to analyze with `--std=93` as at that time `default` was not a reserved identifier.

closed 2024-04-09 by tgingold