ghdl/ghdl #2640: ghdl with mcode backend crashes on if generate when .ghw output is enabled

Jogt, 2024-04-05:
**Description**
GHDL with the mcode backend crashes while elab-running the MWE, if ghw output is enable with the `--wave=wave.ghw` option.
I could not observe the same crash without the `--wave` option, or with the llvm or gcc backends.

**Expected behaviour**
No crash

**How to reproduce?**

MWE:

```vhd :file: mwe.vhd
entity ent1 is
  generic (
    FOO : integer range 2 to 14 := 11
  );
end entity;

architecture structural of ent1 is
begin
  x: if FOO < 9 generate
  else generate
  end generate;
end architecture structural;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 mwe.vhd
ghdl --elab-run --std=08 ent1 --wave=wave.ghw
```

**Context**
Please, provide the following information:

- OS: Arch-Linux
- Origin:
  - [x] Package manager: `aur: ghdl-mcode-git 5.0.0dev.r9345.g14b411c57-1`
  - This builds ghdl from source. Commit hash 14b411c575f5492cde3bd83f8a5713d231ddc6e7

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 5.0.0-dev (4.0.0.r24.g14b411c57) [Dunoon edition]
Compiled with GNAT Version: 13.2.1 20230801
Target: x86_64-pc-linux-gnu
/home/kadel/workspace/ghdl_bug_report/
Command line:
ghdl --elab-run --std=08 ent1 --wave=wave.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-avhpi.adb:593 access check failed
******************************************************************
```

**Additional context**
`ghdl --version`
```
GHDL 5.0.0-dev (4.0.0.r24.g14b411c57) [Dunoon edition]
 Compiled with GNAT Version: 13.2.1 20230801
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

tgingold, 2024-04-07:
I haven't checked, but I am pretty sure it works ok with `--fst=`.

Jogt, 2024-04-08:
Yes, there is no crash if `--fst` export is used

tgingold, 2024-04-09:
I am closing this ticket, I think the issue was already reported.

Xiretza, 2024-04-09:
I can't find any related existing issues.

tgingold, 2024-04-09:
Yep, sorry.

sideprojectslab, 2025-01-19:
+1 for this issue, I have the exact same problem which indeed arises from some generate statements. Is there anything I can do to help resolve this? Further testing or even looking at the code (with some very good pointers from the developers)?

tgingold, 2025-01-22:
There is already `src/simul/simul-fst.adb` to generate fst files (for jit backends).  The ghw dumper should be similarly rewritten (and ideally extended for new features).

closed 2026-09-05 by tgingold