ghdl/ghdl #2641: 4.0.0: issue880 and ticket24 tests fail on ppc64el with the llvm backend

doko42, 2024-04-07:
these tests fail with 4.0.0 on powerpc64le-linux-gnu with GCC 13.2 and LLVM 18.1.2, but succeed when using the GCC backend. This is a regression from 3.0.0.

The tests succeed on x86_64-linux-gnu and aarch64-linux-gnu.

1406s gna issue880: [31mfailed[0m
1406s analyze psl.vhdl
1406s elaborate psl
1406s simulate psl --psl-report=psl.out ( psl --psl-report=psl.out)
1406s 21c21
1406s <    "started-count": 3,
1406s ---
1406s >    "started-count": 8,
1406s report mismatch
1406s 
1406s 
1406s gna ticket24: [31mfailed[0m
1406s analyze psl.vhdl
1406s elaborate psl
1406s simulate psl --psl-report=psl.out ( psl --psl-report=psl.out)
1406s 21c21
1406s <    "started-count": 3,
1406s ---
1406s >    "started-count": 8,
1406s report mismatch
1406s 
1406s 
1406s gna test failed (issue880 ticket24 )

tgingold, 2024-04-10:
I am not sure what I can do as I don't have access to a ppc64 machine.

doko42, 2024-04-10:
such machines exist on the GCC build farm, and you also can use them in Launchpad for PPA builds

commit 76e46088 (2024-04-16): trans-chap9: initialize all the new state vector

Fix #2642
Fix #2641
closed 2024-04-16 by tgingold