ghdl/ghdl #2642: 4.0.0: issue880 and ticket24 tests fail on amd64 with the mcode backend

doko42, 2024-04-07:
these tests fail with 4.0.0 on x86_64-linux-gnu with GCC 13.2 and LLVM 18.1.2, but succeed when using the LLVM or GCC backends. This is a regression from 3.0.0.

3823s gna issue880: [31mfailed[0m
3823s analyze psl.vhdl
3823s elaborate psl
3823s simulate psl --psl-report=psl.out ( psl --psl-report=psl.out)
3823s 21c21
3823s <    "started-count": 3,
3823s ---
3823s >    "started-count": 8,
3823s report mismatch
3823s 
3823s 
3823s gna ticket24: [31mfailed[0m
3823s analyze psl.vhdl
3823s elaborate psl
3823s simulate psl --psl-report=psl.out ( psl --psl-report=psl.out)
3823s 21c21
3823s <    "started-count": 3,
3823s ---
3823s >    "started-count": 8,
3823s report mismatch
3823s 
3823s 
3823s gna test failed (issue880 ticket24 )

tgingold, 2024-04-08:
I have seen them failing from time to time but I haven't investigated the reason.
Not a stopper.

tgingold, 2024-04-10:
It would be nice to have a reliable way to reproduce.  The test doesn't fail in CI (both with mcode or llvm18 on Windows).

tgingold, 2024-04-15:
Ok, I can reproduce the failure with llvm on ppc64el (on cfarm29)

doko42, 2024-04-17:
I can confirm this is fixed. will there be a 4.0.1 release?

tgingold, 2024-04-17:
Great.  Thank you for the status update.

I have already released 4.1 (without the fix).  There is currently no plan for a new 4.x.y release.

commit 76e46088 (2024-04-16): trans-chap9: initialize all the new state vector

Fix #2642
Fix #2641