ghdl/ghdl #2644: UVVM Axistream

erick166, 2024-04-08:
**Description**
GHDL crashes with the following error:

**Expected behaviour**
What you expected to happen, and what is happening instead.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2024-04-21:
I can reproduce on Windows but not on Linux.  I am investigating.

erick166, 2024-04-22:
Alright, I guess this is sort of a temporary solution?
Will you make a fix in a new release or do you suggest that UVVM be optimized in some way?

tgingold, 2024-04-22:
Not so temporary.
I keep that in mind.  I need to rework how memory is allocated and avoid stack allocation for big blocks.
I doubt that UVVM can work around this issue.

helilidada, 2024-04-27:
[image]
can any one fix it?
I tried to debug VTM(VVC 10.0) on MACOS, but it gonna this issue..I did not download YUV file yet.

erick166, 2024-07-18:
A workaround in UVVM to fix this issue has been released in v2 2024.07.03.
The constant C_MAX_TUSER_BITS in axistream_bfm_pkg has been reduced from 64 to 32, and moved to adaptations_pkg to be configurable by the user.

helilidada, 2024-07-18:
thx! bro.

istvan-t-kiss, 2024-07-30:
I think this has been solved in the latest UVVM release [v2024.07.03b](https://github.com/UVVM/UVVM/releases/tag/v2024.07.03b).
For me, it seems to work in a simple TestBench.

tgingold, 2024-07-30:
Great to know!

Paebbels, 2024-12-19:
@tgingold can we close this issue as solved?