ghdl/ghdl #2645: std.textio.read assertion failure for boolean values

erick166, 2024-04-09:
**Description**
When using std.textio.write and std.textio.read with a boolean value, the following GHDL assertion is reported:
```
../../src/std/v08/textio-body.vhdl:735:5:@0ms:(assertion failure): boolean read failure
ghdl:error: assertion failed
ghdl:error: simulation failed
Error: Program ended with exit code 1
```

**Expected behaviour**
The boolean value written to the file is lowercase and therefore it fails, but if it's for instance True or TRUE it doesn't fail.

**How to reproduce?**
Run the following test:

```vhd :file: example_tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

entity example_tb is
end entity;

architecture rtl of example_tb is
begin

  p_main : process
    file file_handler  : text;
    variable v_line    : line;
    variable v_boolean : boolean := True;
  begin
    file_open(file_handler, "output.txt", write_mode);
    write(v_line, v_boolean);
    writeline(file_handler, v_line);
    file_close(file_handler);
    v_line := null;

    file_open(file_handler, "output.txt", read_mode);
    readline(file_handler, v_line);
    read(v_line, v_boolean);
    file_close(file_handler);

    wait;
  end process p_main;

end architecture rtl;
```

**Context**
- OS: Windows 11
- GHDL 5.0.0-dev (4.0.0.r24.g14b411c57)
- UVVM v2 2024.03.08

erick166, 2024-04-09:
I found this issue which is related to this problem: https://github.com/ghdl/ghdl/issues/2115

tgingold, 2024-04-09:
Indeed.  I will push a fix soon