ghdl/ghdl #2646: Overflow in exponentiation

erick166, 2024-04-09:
**Description**
I get an overflow error and I wonder if it's a GHDL or the code problem.

**How to reproduce?**
The testbench below has 3 scenarios:
First, using the attribute to print the maximum integer positive value which works okay.
Then, using exponentiation, which works for the to_string() procedure but not for the other 2 cases, which give the following error:
```
ghdl:error: overflow in exponentiation
ghdl:error: simulation failed
Error: Program ended with exit code 1
```
Finally, using exponentiation but with a hardcoded value, which gives the following errors:
```
example_tb.vhd:24:46:warning: arithmetic overflow in static expression [-Wruntime-error]
report "Using 'image:" & integer'image(2 ** 31 - 1);
^
example_tb.vhd:25:21:warning: arithmetic overflow in static expression [-Wruntime-error]
v_integer := (2 ** 31) - 1;       
^      
```                           

```vhd :file: example_tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity example_tb is
end entity;

architecture rtl of example_tb is
begin

  p_main : process
    variable v_exponent : integer := 31;
    variable v_integer  : integer;
  begin
    report "Using to_string:" & to_string(integer'right);
    report "Using 'image:" & integer'image(integer'right);
    v_integer := integer'right;

    report "Using to_string:" & to_string(2 ** v_exponent - 1);
    report "Using 'image:" & integer'image(2 ** v_exponent - 1);
    v_integer := (2 ** v_exponent) - 1;

    --report "Using to_string:" & to_string(2 ** 31 - 1);
    --report "Using 'image:" & integer'image(2 ** 31 - 1);
    --v_integer := (2 ** 31) - 1;

    wait;
  end process p_main;

end architecture rtl;
```

**Context**
- OS: Windows 11
- GHDL 5.0.0-dev (4.0.0.r24.g14b411c57)
- UVVM v2 2024.03.08

tgingold, 2024-04-10:
I think this is correct. 2**31 is beyond the bounds of integer.

erick166, 2024-04-11:
But since I'm actually using (2 ** 31 - 1) perhaps it should be okay? I just wonder because if it's used inside the to_string() function it works fine.

tgingold, 2024-04-20:
So 2**31 is out of the integer range, that's why you get overflow in most cases.

For `to_integer(2**v_exponent - 1)`, there are many possible types, and according to vhdl it should be universal_integer, which is a 64b type in ghdl.  That's why there is no overflow in that case.

erick166, 2024-04-22:
Right, I just thought since I'm actually using (2 ** 31 - 1) and not 2 ** 31 as the integer value perhaps it could work, since it works when using it inside the to_string() function, but if it's not the case then I'll just use a workaround with integer'right.

nselvara, 2024-04-22:
I guess the `**` operator is evaluated first as it has higher precedence which then leads to overflow of the value.
You can use `integer'high` which should return you always the highest value.

commit e57c785f (2024-04-27): testsuite/gna: add a test for #2646

Close #2646
closed 2024-04-25 by erick166