ghdl/ghdl #2647: Non-synthesizable code leads to latches

tristanitschner, 2024-04-11:
**Description**

Non-synthesizable code can lead to latches in yosys synthesis. In most cases, this is not a problem, as yosys optimizes the latches away. However, I have synthesized a large design, in which this is not the case. This lead to an error, as the target architecture does not support latches. I cannot post the large design, so I made a small example:

```vhdl
library ieee; 
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity case_statement_not_full is
	port(
        x : in  std_logic_vector(1 downto 0);
        y : out std_logic_vector(1 downto 0)
	);
end entity;

architecture arch of case_statement_not_full is
begin

    process (all) is
    begin
        case x is
            when "00" => y <= "00";
            when "01" => y <= "01";
            when "10" => y <= "10";
            when "11" => y <= "11";
            when others => null;
        end case;
    end process;

end architecture;

```

```vhdl
library ieee; 
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity case_statement_full is
	port(
        x : in  std_logic_vector(1 downto 0);
        y : out std_logic_vector(1 downto 0)
	);
end entity;

architecture arch of case_statement_full is
begin

    process (all) is
    begin
        case x is
            when "00" => y <= "00";
            when "01" => y <= "01";
            when "10" => y <= "10";
            when "11" => y <= "11";
            when others => y <= "XX";
        end case;
    end process;

end architecture;
```

I also repeated the same experiment with verilog:

```verilog
module case_statement_full (
    input  wire [1:0] x,
    output reg  [1:0] y
);

always @*
    case (x)
        2'b00: y = 2'b00;
        2'b01: y = 2'b01;
        2'b10: y = 2'b10;
        2'b11: y = 2'b11;
        default: y = 2'bxx;
    endcase

endmodule
```

```verilog
module case_statement_not_full (
    input  wire [1:0] x,
    output reg  [1:0] y
);

always @*
    case (x)
        2'b00: y = 2'b00;
        2'b01: y = 2'b01;
        2'b10: y = 2'b10;
        2'b11: y = 2'b11;
        default: ;
    endcase

endmodule

```

For yosys synthesis, I took two paths: Once using ghdl synth, the other using ghdl-yosys-plugin. For convenience sake, here is a Makefile which produces all testcases:

```Makefile
BUILD := build

VEROUT := $(BUILD)/verilog
VHDOUT := $(BUILD)/vhdl
GHDLVERSYNTHOUT := $(BUILD)/ver_synth
GHDLVHDSYNTHOUT := $(BUILD)/vhd_synth

GHDL := ghdl
GHDLWORKDIR := $(BUILD)/ghdlworkdir
GHDLFLAGS := --std=08 --workdir=$(GHDLWORKDIR)

TARGETS += $(VEROUT)/case_statement_full.json
TARGETS += $(VEROUT)/case_statement_not_full.json
TARGETS += $(VHDOUT)/case_statement_full.json
TARGETS += $(VHDOUT)/case_statement_not_full.json
TARGETS += $(GHDLVERSYNTHOUT)/case_statement_full.json
TARGETS += $(GHDLVERSYNTHOUT)/case_statement_not_full.json
TARGETS += $(GHDLVHDSYNTHOUT)/case_statement_full.json
TARGETS += $(GHDLVHDSYNTHOUT)/case_statement_not_full.json

all: $(TARGETS)

$(VHDOUT)/%.json: %.vhd | $(VHDOUT) $(GHDLWORKDIR)
	$(GHDL) -a $(GHDLFLAGS) $<
	yosys -l $(VHDOUT)/$<.log -m ghdl -p 'ghdl $(GHDLFLAGS) $(basename $<); check; synth_ecp5; write_json $@'

$(VEROUT)/%.json: %.v | $(VEROUT)
	yosys -l $(VEROUT)/$<.log -m ghdl -p 'read_verilog $<; check; synth_ecp5; write_json $@'

$(GHDLVERSYNTHOUT)/%.v: %.vhd | $(GHDLVERSYNTHOUT) $(GHDLWORKDIR)
	$(GHDL) -a $(GHDLFLAGS) $<
	$(GHDL) --synth $(GHDLFLAGS) --out=verilog $(basename $<) > $@

$(GHDLVERSYNTHOUT)/%.json: $(GHDLVERSYNTHOUT)/%.v | $(GHDLVERSYNTHOUT)
	yosys -l $<.log -p 'read_verilog $<; check; synth_ecp5; write_json $@'

.PRECIOUS: $(GHDLVERSYNTHOUT)/%.vhd
$(GHDLVHDSYNTHOUT)/%.vhd: %.vhd | $(GHDLVHDSYNTHOUT) $(GHDLWORKDIR)
	$(GHDL) -a $(GHDLFLAGS) $<
	$(GHDL) --synth $(GHDLFLAGS) --out=vhdl $(basename $<) > $@

.PRECIOUS: $(GHDLVHDSYNTHOUT)/%.vhd
$(GHDLVHDSYNTHOUT)/%.json: $(GHDLVHDSYNTHOUT)/%.vhd | $(GHDLVERSYNTHOUT)
	$(GHDL) -a $(GHDLFLAGS) $<
	yosys -l $<.log -m ghdl -p 'ghdl $(GHDLFLAGS) $(basename $(notdir $<)); check; synth_ecp5; write_json $@'

$(BUILD):
	mkdir -p $@

$(VEROUT):
	mkdir -p $@

$(VHDOUT):
	mkdir -p $@

$(GHDLVERSYNTHOUT):
	mkdir -p $@

$(GHDLVHDSYNTHOUT):
	mkdir -p $@

$(GHDLWORKDIR):
	mkdir -p $@

.PHONY: clean
clean:
	$(RM) -r $(BUILD)

```

The latches occur in both ghdl synthesis cases. See build/vhd_synth/case_statement_not_full.vhd.log and build/vhdl/case_statement_not_full.vhd.log. They do, however, not occur in build/ver_synth/case_statement_not_full.v.log as well as the other testcases.

**Expected behaviour**

In fact, the reason for this could be the VHDL standard, I have not checked. I that case, I'd suggest adding fixed behavior to -frelaxed or -frelaxed-rules.

tristanitschner, 2024-04-11:
PS: Of course, the line
```
.PRECIOUS: $(GHDLVERSYNTHOUT)/%.vhd
```
in the Makefile should read
```
.PRECIOUS: $(GHDLVERSYNTHOUT)/%.v
```

tgingold, 2024-04-14:
I am not sure to understand the issue.  What is the point if the code is not synthesizable ?

tristanitschner, 2024-04-14:
If the code is not synthesizable, it should not create latches, or if it does, ghdl should warn about it. It's just that I've seen this pattern in others code very often: a fully combinatorial case state with missing assignment for default case. Not that the code is actually not synthesizable, but the non-synthesizable expression for the default case "<= (others => 'X')" fixes the issue, i.e. leads to no latches being inferred. As can be seen by the verilog example, with the yosys' verilog backend, this is not the case in the first place at all. I think ghdl's behaviour should match.

tgingold, 2024-04-17:
Ok.  Latches are detected when created by if statement, but not yet by case statement.  I am investigating.

commit 6ab71231 (2024-04-19): netlists-inference: detect simple logical loop in case statement

Fix #2647
closed 2024-04-19 by tgingold