ghdl/ghdl #2649: 4.1.0: undefined reference to `llvm::EngineBuilder::EngineBuilder ...)`

yurivict, 2024-04-15:
**Description**
Link fails:
```
gnatlink ortho_code_main.ali -o ghdl1-llvm -g llvm-cbindings.o grt-cstdio.o grt-cdynload.o --LINK=c++ -L/usr/local/llvm15/lib -lLLVM-15
/usr/local/bin/ld: llvm-cbindings.o: in function `llvm_jit_init':
/usr/ports/cad/ghdl/work/ghdl-4.1.0/src/ortho/llvm6/llvm-cbindings.cpp:2890: undefined reference to `llvm::EngineBuilder::EngineBuilder(std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >)'
collect2: error: ld returned 1 exit status
gnatlink: error when calling /usr/local/gnat12/bin/c++
```

**How to reproduce?**
Regular build.

**Context**
Problem was discovered during build of the FreeBSD port cad/ghdl.

- OS: FreeBSD 14.0
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 4.1.0

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

tgingold, 2024-04-16:
This looks like particular to FreeBSD (or more probably how llvm was configured on FreeBSD).
You can safely #if 0/#endif the llvm_jit functions at the end of llvm-cbinding.cpp

What is the output of `llvm-config --components` ?

yurivict, 2024-04-16:
$ llvm-config15 --components 
aarch64 aarch64asmparser aarch64codegen aarch64desc aarch64disassembler aarch64info aarch64utils aggressiveinstcombine all all-targets amdgpu amdgpuasmparser amdgpucodegen amdgpudesc amdgpudisassembler amdgpuinfo amdgputargetmca amdgpuutils analysis arm armasmparser armcodegen armdesc armdisassembler arminfo armutils asmparser asmprinter avr avrasmparser avrcodegen avrdesc avrdisassembler avrinfo binaryformat bitreader bitstreamreader bitwriter bpf bpfasmparser bpfcodegen bpfdesc bpfdisassembler bpfinfo cfguard codegen core coroutines coverage debuginfocodeview debuginfodwarf debuginfogsym debuginfomsf debuginfopdb demangle dlltooldriver dwarflinker dwp engine executionengine extensions filecheck frontendopenacc frontendopenmp fuzzercli fuzzmutate globalisel hexagon hexagonasmparser hexagoncodegen hexagondesc hexagondisassembler hexagoninfo instcombine instrumentation interfacestub interpreter ipo irreader jitlink lanai lanaiasmparser lanaicodegen lanaidesc lanaidisassembler lanaiinfo libdriver lineeditor linker lto mc mca mcdisassembler mcjit mcparser mips mipsasmparser mipscodegen mipsdesc mipsdisassembler mipsinfo mirparser msp430 msp430asmparser msp430codegen msp430desc msp430disassembler msp430info native nativecodegen nvptx nvptxcodegen nvptxdesc nvptxinfo objcarcopts objcopy object objectyaml option orcjit orcshared orctargetprocess passes powerpc powerpcasmparser powerpccodegen powerpcdesc powerpcdisassembler powerpcinfo profiledata remarks riscv riscvasmparser riscvcodegen riscvdesc riscvdisassembler riscvinfo runtimedyld scalaropts selectiondag sparc sparcasmparser sparccodegen sparcdesc sparcdisassembler sparcinfo support symbolize systemz systemzasmparser systemzcodegen systemzdesc systemzdisassembler systemzinfo tablegen target textapi transformutils ve veasmparser vecodegen vectorize vedesc vedisassembler veinfo webassembly webassemblyasmparser webassemblycodegen webassemblydesc webassemblydisassembler webassemblyinfo webassemblyutils windowsdriver windowsmanifest x86 x86asmparser x86codegen x86desc x86disassembler x86info x86targetmca xcore xcorecodegen xcoredesc xcoredisassembler xcoreinfo xray

yurivict, 2024-04-16:
I tried to compile and link this code:
```
#include "llvm-c/Target.h"
#include "llvm/IR/Type.h"
#include "llvm/IR/Value.h"
#include "llvm/IR/LLVMContext.h"
#include "llvm/Config/llvm-config.h"
#include "llvm-c/TargetMachine.h"
#include "llvm-c/Core.h"
#include "llvm-c/BitWriter.h"

#include "llvm-c/Analysis.h"

#include <llvm/ExecutionEngine/ExecutionEngine.h>
#include <llvm/ExecutionEngine/MCJIT.h>
#include <llvm/IR/Mangler.h>

using namespace llvm;

static LLVMModuleRef TheModule;
static ExecutionEngine *EE;

int main() {
  std::string error;

  EE = llvm::EngineBuilder(std::unique_ptr<Module>(unwrap(TheModule)))
    .setErrorStr(&error)
    .setEngineKind(llvm::EngineKind::JIT)
     .create();
}
```
but it links fine.

tgingold, 2024-04-17:
Weird.  Two questions:
how do you compile and link your small reproducer ?
Is it exactly the same symbol (not demangled, as reported by nm) ?

yurivict, 2024-05-25:
The reason is in symbol name mismatch.

The symbol exported from libLLVM-15.so is
```_ZN4llvm13EngineBuilderC1ENSt3__110unique_ptrINS_6ModuleENS1_14default_deleteIS3_EEEE```
which is same as
```_ZN4llvm13EngineBuilderC1ENSt3__110unique_ptrINS_6ModuleENS1_14default_deleteIS3_EEEE```

The symbol expected by llvm-cbindings.o is
```_ZN4llvm13EngineBuilderC1ESt10unique_ptrINS_6ModuleESt14default_deleteIS2_EE```
which is same as
```llvm::EngineBuilder::EngineBuilder(std::unique_ptr<llvm::Module, std::default_delete<llvm::Module>>)```.

```__1``` comes from [C++11 inline namespaces](https://stackoverflow.com/questions/11016220/what-are-inline-namespaces-for).

It turned out that llvm15 is compiled partially with C++14, and partially with C++11.

Does ghdl support more modern LLVM versions?

tgingold, 2024-05-25:
Yes, I have tested at least llvm 14 and llvm 17.

yurivict, 2024-05-25:
I figured out what was the problem: it was built using libLLVM-15.so by the mismatching clang++ compiler from llvm-18.

Thank you for your help!

closed 2024-05-25 by yurivict