ghdl/ghdl #2650: Type of element is ambigous in a nested array or record position asignment with concatenation

EmilioPeJu, 2024-04-17:
**Description**
If an element inside a array or record is set using concatenation and position assignment, the type becomes ambiguous. Removing the concatenation or changing the position assignment to specify the position, makes it work.

**Expected behaviour**
The type should be inferred even if the value is a concatenation

**How to reproduce?**
Here is a minimum example:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity testbench is
end;

architecture arch of testbench is
    type vector_array is array(natural range <>) of std_ulogic_vector;
    signal a : vector_array(0 to 1)(15 downto 0);
begin
    a <= (X"12" & X"34", X"56" & X"78");            -- Should work
    --a <= (0 => X"12" & X"34", 1 => X"56" & X"78");  -- Workaround
end;
```

**Context**
Please, provide the following information:

- OS: Red Hat Enterprise Linux release 8.9 (Ootpa)
- Origin:
  - [x] Package manager: nix (Nix) 2.18.1
  - [x] Built from sources: `commit 76e460880c31b771f082920a4fbefc47fd6a0485`

Build recipe was slightly changed to use latest commit(at the time) in the repository.
Here is the version information from the command:
```bash
❯ ghdl --version
GHDL 5.0.0-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 12.3.0
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2024 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

suryabvsp, 2024-04-18:
Seems similar to the issue I'm facing currently with the latest GHDL (4.1.0 (4.0.0.r39.g7188e92cf)) compiled with LLVM on Ubuntu 20.04.

I get `error: type of element is ambiguous` when an array of std_logic_vectors is defined with concatenations present in its elements:
```
< NOTE: MAX_X and MAX_Y are generics to the entity, defined as positive with values 400. >
type vector_array_10len_32bit is array (0 to 9) of std_logic_vector(31 downto 0);
constant CFG_DATA : vector_array_10len_32bit := (
        (std_logic_vector(to_unsigned(MAX_Y, 16)) & std_logic_vector(to_unsigned(MAX_X, 16))),
        (x"00000000"),
        (x"00000000"),
        (x"00000000"),
        (x"00000000"),
        (x"00000000"),
        (x"00000000"),
        (x"00000000"),
        (x"00000000"),
        (x"00000000")
);
```
I was able to circumvent the error by changing the the element definition to `std_logic_vector((to_unsigned(MAX_Y, 16)) & (to_unsigned(MAX_X, 16)))`. 

However, this issue does not come up in Questasim/Vivado and hence a fix would be greatly appreciated to help us avoid changes to our existing source codes. Thank you.

tgingold, 2024-04-18:
I fear that ghdl is correct here, but I need to double check with the LRM.  This is due to a new feature of vhdl-2008 (and doesn't appear with previous vhdl versions).

tgingold, 2024-04-19:
I think that ghdl is correct according to LRM 9.3.3.3 array types:
```
For an aggregate of a one-dimensional array type, each choice shall specify values of the index type, and the
expression of each element association shall be of either the element type or the type of the aggregate.
```

As `X"12" & X"34"` can be either an `std_logic_vector` or an array of `std_logic_vector` (whose type is `vector_array`), the expression is ambiguous.

Now, if there is a compatibility problem with other tools, I can add a rule (under `-frelax`) to fix the issue.

suryabvsp, 2024-04-19:
Yeah that should be quite helpful. I assume once that's done we'd just need to add a flag to our ghdl compilation options?

Araneidae, 2024-04-19:
I didn't realise you can concatenate vectors to form higher dimensional vectors!

Please, if you're happy to use `-frelax` as a guard for unofficial language extensions that are supported elsewhere, please let this work.  I also work with the intersection of 2008∩Vivado∩ModelSim, and was dismayed at how pendantic ghdl is!

Does `-frelax` already exist, or is this going to have to be a new feature?  There are a number of issues with the interpretation of `others` which I'll raise if necessary.  Also, to my surprise, ModelSim let me use selected assignment in a sequential context which ghdl didn't like, is it worth flagging that?  (Hadn't gotten around to discovering if Vivado supports that.)

tgingold, 2024-04-20:
So, the option `-frelaxed` now makes the code compiled.  I am not sure about all the consequences, so I am interested in feedback.

@Araneidae yes please, if you know some issues about compatibility with other simulators, please open issues!

EmilioPeJu, 2024-04-22:
Thanks a lot for providing the `-frelaxed` option, it is much appreciated.

tgingold, 2024-04-22:
You're welcome!

suryabvsp, 2024-04-23:
Sorry for my late reply, I tested the latest nightly build on my Ubuntu pc with the above code. I use the following flags (I use vunit via python, but it invokes the ghdl flags accordingly):
```python
vu.set_sim_option(
        "ghdl.sim_flags", [f"--stack-max-size={stack_size}", f"--max-stack-alloc=0"]
    )
vu.add_compile_option("ghdl.a_flags", [f"-frelaxed"])```
Do you see any change in the way I should use frelaxed?

suryabvsp, 2024-04-25:
No success after adding `-frelaxed` to both sim and compilation flags.

The commands being executed (by vunit) are: 

Compilation command (-a flag):
```bash
/home/my_user/ghdl_llvm_nightly_apr20/bin/ghdl -a --workdir=/home/my_user/my_ip/sim/custom_design_tb/ghdl/libraries/custom_design_tb --work=custom_design_tb --std=08 -P/home/my_user/my_ip/sim/custom_design_tb/ghdl/libraries/vunit_lib -P/home/my_user/my_ip/sim/custom_design_tb/ghdl/libraries/osvvm -P/home/my_user/my_ip/sim/custom_design_tb/ghdl/libraries/custom_design_tb -frelaxed /home/my_user/my_ip/sim/tb/custom_design_tb.vhd
```
Simulation command (--elab-run flag):
```bash
/home/my_user/ghdl_llvm_nightly_apr20/bin/ghdl --elab-run --std=08 --work=custom_design_tb --workdir=/home/my_user/my_ip/ghdl/libraries/custom_design_tb -P/home/my_user/my_ip/ghdl/libraries/vunit_lib -P/home/my_user/my_ip/ghdl/libraries/osvvm -P/home/my_user/my_ip/ghdl/libraries/custom_design_tb -o /home/my_user/my_ip/test_output/custom_design_tb.custom_design_tb.counter_op.custom_design_test_b141/ghdl/custom_design_tb-tb custom_design_tb tb -frelaxed -gtb_path=/home/my_user/my_ip/sim/tb/ -gtb_cfg=test_name:counter_op -goutput_path=/home/my_user/my_ip/test_output/custom_design_tb.custom_design_tb.counter_op.custom_design_test_b141/ -grunner_cfg=active python runner : true,enabled_test_cases : custom_design_test,output path : /home/my_user/my_ip/test_output/custom_design_tb.custom_design_tb.counter_op.custom_design_test_b141/,tb path : /home/my_user/my_ip/sim/tb/,use_color : true --assert-level=error
```

Does `-frelaxed` need to be set under a different flag?

tgingold, 2024-04-25:
@suryabvsp The flag is present, but are you using a recent version of ghdl which includes this fix ?

suryabvsp, 2024-04-26:
I used the [nightly build released after your fix](https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-llvm.tgz)

suryabvsp, 2024-04-29:
More details about my run:
ghdl version:
[image]

error message :
[image]

which mentions the same ghdl path as the one mentioned at:
[image]

tgingold, 2024-04-29:
@suryabvsp Please provide a complete reproducer so that I can investigate.

suryabvsp, 2024-04-29:
I was able to use your fix successfully. Turns out I may have been using the wrong flags.
What did not work :
```python
    stack_size = "1000mb"
    vu.set_sim_option(
        "ghdl.sim_flags", [f"--stack-max-size={stack_size}", f"--max-stack-alloc=0"]
    )
    vu.add_compile_option("ghdl.a_flags", [f"-frelaxed"])
```
What finally worked:
```python
    stack_size = "1000mb"
    vu.set_compile_option("ghdl.a_flags", ["-frelaxed"])
    vu.set_sim_option("ghdl.elab_flags", ["-frelaxed"])
```
Thanks a lot for the help and patience!

tgingold, 2024-04-30:
Great!  Thank you for the update.

commit ae75c96f (2024-04-19): sem_array_aggregate_elements: use -frelaxed

If the element is not a scalar, -frelaxed does not allow
element of the aggregate type.

Fix #2650
closed 2024-04-30 by tgingold